nefarius/DsHidMini #1: Not being picked up by DS4Windows v2.2.9

Skynet19972029, 2021-03-07:
Hello Nefarious! Thank you so much for this tool.
Been testing it so far only with DirectInput and RCPS3, with both working great.
Need to still test the PCSX2 mode.

But, with the DS4 mode, the pad is not being detected at all in DS4Windows.
A real DS4 has no problem connecting to it.
But the DS3 with the DS4 mode it's not dice.

Anything (log or else) I have to send you for you to debug?

Thank you.

[image]

Kanuan, 2021-03-07:
Two things happening right now:
 1. The current DsHidMini driver has a bug in DS4 mode that is mixing buttons/triggers, so you won't be able to use the controller correctly for now.
 2. DS3 support on DS4Windows is only working on the Beta Build for now;

If you want to test anyway, download the Beta Build from here [https://github.com/mika-n/DS4Windows/raw/debugVerWPF/extras/DS4WindowsWPF_DeviceDetectDebug.zip] and place it at the same folder as the normal DS4windows.exe, then execute the DS4WindowsDebug.exe (make sure the normal DS4windows.exe is not running).

 Keep an eye on the DsHidMini release page to see when the fixed version is out

nefarius, 2021-03-07:
Yep, you people are fast as lightning, we didn't have the time to remark that it's not in the latest DS4Windows release yet, you need the mentioned Beta build 😉 

Button mappings are wrong/broken though, we're already aware and working on it.

Stay tuned!

mika-n, 2021-03-07:
@Skynet19972029 The logfile screenshot seem to indicate that you still have an old 1.16.112 version of ViGem virtual gamepad driver. It is strongly recommended to use the latest 1.17.x version. DS4Windows auto-update feature automagically updates only the DS4Windows app itself, but not the ViGem bus driver. To update that you should head to the ViGemBus site maintained by Nefarius and to download the latest ViGem installer from there.
https://github.com/ViGEm/ViGEmBus/releases

Skynet19972029, 2021-03-07:
Ah, nice. 
Thank you all so much.

I JUST wrote a new ticket issue on PCSX2, I hope it's not related to all the other stuff.

@mika-n 
I SWORE I had the latest ViGem driver. I think the issue was that when I installed Parsec it overwrote it.
Thanks for noticing!

nefarius, 2021-03-07:
I'll leave it open until new fixed release is ready, so other may find it in the meantime.

nefarius, 2021-03-08:
Fixed release online!

closed 2021-03-08 by nefarius