nefarius/DsHidMini #2: PCSX2 and pressure sensitive buttons.

Skynet19972029, 2021-03-07:
I just finished testing the PCSX2 mode.

There are a couple of problems I encountered unless I have done something wrong.

When enabling the PCSX2 mode, the pad work perfectly in PCSX2 (V1.6.0 Stable), can map all the buttons, but, no pressure sensitive buttons.
When you enable the "Mute digital pressure buttons" Now, the face buttons and shoulder buttons are mapped as sliders, so they are pressure sensitive. But, you can't map Start, Select and both L3 and R3. And also, for some reason the Left stick keeps continuously pushing to the left and can't make it stop, even when setting a high deadzone in PCSX2.
When checking the pad in "joy.cpl" everything is correct regarding the "left stick" pushing left. I noticed that only X and O are sliders there, if that matters.

The weird behavior also occurs in other DirectInput applications.

nefarius, 2021-03-07:
The way you're supposed to map is as follows:

- Enable the Mute option
- Map all axes and buttons with pressure values (Face, Shoulder, DPad, Thumbs, Triggers)
- Disable mute
- Map Select, Start, L3 and R3
- Save
- Done!

Cheers

Skynet19972029, 2021-03-07:
Thank you, nefarius!
I see my error, though, to be fair, I would have never thought on doing it that way!

Have a great day!

Skynet19972029, 2021-03-07:
> The way you're supposed to map is as follows:
> 
> * Enable the Mute option
> * Map all axes and buttons with pressure values (Face, Shoulder, DPad, Thumbs, Triggers)
> * Disable mute
> * Map Select, Start, L3 and R3
> * Save
> * Done!
> 
> Cheers

Skynet19972029, 2021-03-08:
Update.
I did it like that.
But, I think it's the left stick, keeps pushing down and can't navigate. Though, in joy.cpl it's all correct.
[image]

https://user-images.githubusercontent.com/72090667/110260764-4f72ba80-7f8c-11eb-93f7-dcfdcfeebccc.mp4

nefarius, 2021-03-08:
@Skynet19972029 strange 🤔 what game is that? [I made a video walking through the configuration](https://www.youtube.com/watch?v=CBuCdEWhQEc) and can't really see any issue or replicate it myself.

Cheers

Skynet19972029, 2021-03-08:
Metal Gear Solid 3 Substance.
This behavior is the same in all games sadly.
I will redo the config just to retry following your video just in case I did something wrong.

nefarius, 2021-03-08:
> Metal Gear Solid 3 Substance.

Let me, uh, quickly look in my cabinet for the legit copy I'm certain I own and test it myself 😇

Skynet19972029, 2021-03-08:
Ok.
After fiddling around for a while. I decided to redo everything. Starting from the drivers.
I followed you step by step on the video and doing all of it, fixed the issue! So, there must have been something wrong with the drivers perhaps?
Regardless, it's working now! Thank you so much for everything!

This is unrelated to the issue I had. But, I was wondering, it's OK that my pad is recognized as "PS3 GamePad" in joy.cpl and not PLAYSTATION(R) 3 Controller as yours?

nefarius, 2021-03-08:
> This is unrelated to the issue I had. But, I was wondering, it's OK that my pad is recognized as "PS3 GamePad" in joy.cpl and not PLAYSTATION(R) 3 Controller as yours?

That means you have an aftermarket controller, the name you see there is taken directly from the device itself, `PLAYSTATION(R) 3 Controller` is the one and only name used by official Sony hardware and aftermarket devices vary it to fly under the trademark violation radar.

Glad to hear it's working!

Cheers

Skynet19972029, 2021-03-08:
But the weird thing is that aftermarket controllers do not have sixaxis nor preassure sensitive buttons at all. 
Which this one does have. And the analog sticks are exactly the same as the ones in the original controller. It also has the correct serial number. As this pad was taken from a used and dead PS3.
Could it be that it says it like that because it's from Mexico? As this controller is for the South America region but they came from Mexico.
I did have connected in this windows install an actual aftermarket pad. Could it be a driver thing?

nefarius, 2021-03-08:
> But the weird thing is that aftermarket controllers do not have sixaxis nor preassure sensitive buttons at all.

That is not true, it greatly varies what aftermarket devices look like and support in features.

> Which this one does have. And the analog sticks are exactly the same as the ones in the original controller. It also has the correct serial number. As this pad was taken from a used and dead PS3.

[I've explained that countless times already](https://vigem.org/projects/BthPS3/Frequently-Asked-Questions/#what-controllers-are-supported); it is not possible to a 100% be certain if it's Sony hardware or not. If it works for you; great! Don't think too much about it! 😁 

> I did have connected in this windows install an actual aftermarket pad. Could it be a driver thing?

I don't quite understand what you mean but no driver has any influence on how the controller firmware/ASICs work.

Skynet19972029, 2021-03-08:
I see.
Though, I won't stress about it. I know that the pad come out of a sealed box form a now broken PS3 and it works with the software just fine.

Thank you once again, nefarius!

nefarius, 2021-03-08:
> I see.
> Though, I won't stress about it. I know that the pad come out of a sealed box form a now broken PS3 and it works with the software just fine.
> 
> Thank you once again, nefarius!

Maybe they changed it in later stages of controller production or outsourced manufacturing, we will never know the full story unless some Sony official can enlighten us 🙂 

I'll close the issue.

closed 2021-03-08 by nefarius