nefarius/DsHidMini #4: Rumble not working

kjp949, 2021-03-10:
DsHidMini 1.0.18.0
BthPS3 1.3.127
ViGEmBus 1.17.333
DS4Windows 2.2.10
Genuine DS3
Asus BT400 BT adapter

I've got my controller working almost perfectly with DS4Windows as an emulated xbox 360 controller, however, I cannot get rumble to work.  Is there some hidden setting somewhere I am missing?  Rumble did work quite well with Fireshock/Shibari.

nefarius, 2021-03-10:
Does it work on USB or not on both protocols? We're currently investigating a problem with rumble not working on Bluetooth on some DS3 variants indeed.

kjp949, 2021-03-10:
Just checked and rumble appears to function correctly over USB.

Kanuan, 2021-03-11:
@kjp949 if possible, could you join the discord server to help with some tests? Nefarius is currently building some beta builds of dshidmini to try to identify the issue, though it would require you to use windows in test-mode to test them

nefarius, 2021-03-11:
# Troubleshooting instructions

- Grab this tool and extract it somewhere: [MightyHIDTest.zip](https://github.com/ViGEm/DsHidMini/files/6125409/MightyHIDTest.zip)
- Connect the DS3 **via Bluetooth**, make sure it's in **SXS mode**
- Run the `MightyHIDTest.exe`
  - It will attempt to send the report defined in `settings.json` to the controller, by default it will just try to set LED 1 lit
- Change to this report in `settings.json` to see if you can make the small motor rumble:  
`0100000000FE01FF000000000002FF27100032FF27100032FF27100032FF27100032000000000000000000000000000000`
- Don't forget to save the file
- Run the tool again
- Small motor should now rumble for a few seconds

kjp949, 2021-03-11:
Ok I ran the MightyHIDTest.  Is the report automatically uploaded somewhere?  I'm not seeing any file created.

Kanuan, 2021-03-11:
running MightyHIDTest should set the controller Led to 1 and make it vibrate for a while. Test both bluetooth/wired and then reply on what happened (make sure the controller is in SXS mode both wired/bluetooth before running the test)

kjp949, 2021-03-11:
Bluetooth: LED #4 is on.  No change in LED while MightyHIDTest is running.  No rumble either.
Console output:
Looking for DsHidMini devices
Found one
Sending report
Done!

USB: LED #1 on while program running.  Vibration motor running as well.  Console output the same as bluetooth.

nefarius, 2021-03-11:
Well, that is not good 😬 did you use Shibari before by any chance? If so, did it work there (on Bluetooth)?

kjp949, 2021-03-11:
Yes rumble worked over Bluetooth with shibari.

mika-n, 2021-03-11:
Please be aware that the latest official version of DS4Windows 2.2.10 doesn't enable rumbling if the ds4windows profile has dualshock4 output mode instead of xbox360 mode. This is intentional because sometimes the rumbling got stuck in "permanent rumbling" state. 

The beta/debug version of DS4Windows app does enable rumble in ds4 output mode also, so you need this if you want to use rumbling with ds4 output controllers.
https://github.com/Ryochan7/DS4Windows/wiki/Troubleshooting#detailed-debug-log-file-from-gamepad-connection-problems

Kanuan, 2021-03-11:
not related in this case. We are testing on SXS mode (RPCS3 mode) and directly sending a custom packet to the controller

nefarius, 2021-03-11:
Yep, this is not a DS4Windows issue but some weird protocol-agnostic problem. Thanks for the hint though.

nefarius, 2021-03-12:
@kjp949 what is your controller model, the value on the sticker on the back?

kjp949, 2021-03-12:
CECHZC2U

Kanuan, 2021-03-12:
For the record, my SIXAXIS controller that has the same behaviour is CECHZ1U

kjp949, 2021-03-12:
No A2 on mine.  Here is the whole label.  Maybe there's some other clue on there.

[image]

nefarius, 2021-03-12:
Could you please test with release `v1.0.32`, thanks.

kjp949, 2021-03-12:
Rumble works with 1.0.32.  Thanks!

closed 2021-03-12 by kjp949

HalasKYO, 2023-03-02:
I have the same issue, earlier worked with SCPToolkit, but because it messed ub bluetooth i jumped on this . if set to SXS, no rumble either usb or bluetooth. if I download the above mentioned file it changes led to 1, rumble, but not in steam or any other app . If I set it to XImput , vibrate works controller CECH ZC2E