nefarius/DsHidMini #12: Windows detecting both DS3 and x360 emulated controller (is there a workaround/fix?)

rjmxtechnology, 2021-03-13:
Is there a way to stop windows from detecting the DS3 at the same time as the emulated 360 when using dshidmini with DS4Windows? My games seem to be showing both the xbox and ds3 inputs when I connect a controller

AshUchiha, 2021-03-13:
Yes tick Hide Ds4 Controller in Ds4Windows settings.

nefarius, 2021-03-13:
Until there's a better solution available check out the previous comment and the documentation provided in regards to setting up HidGuardian.

rjmxtechnology, 2021-03-13:
> 
> 
> Yes, tick (enable) Hide Ds4 Controller in Ds4Windows settings.
Thanks! This seemed to do the trick. For some reason I had missed it when looking through the settings and thought it had been removed in an update or something. Guess I was mistaken. 

> 
> 
> Until there's a better solution available check out the previous comment and the documentation provided in regards to setting up HidGuardian.
I did a bit of a readthrough on hidguardian and I was having trouble grasping how exactly to use it and what all the options mean. Is there a good place for documentation on it?

nefarius, 2021-03-14:
[Same site](https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/#solving-double-input-issues-on-games), mate 😁

rjmxtechnology, 2021-03-14:
Okay thanks will take a look

closed 2021-03-14 by rjmxtechnology