nefarius/DsHidMini #16: Vibration Erratic with latest binaries (but it kinda works)

ferociousdiablo, 2021-03-15:
Hey Nefarius, great to see you working on new stuff again! I'm seeing a problem with the new version, which seems to cause vibration events to persist until the next vibration command is sent or something like that? Generally the vibration seems to be very inconsistent overall. For example shooting an automatic weapon has vibration for the first few shots then nothing. Sometimes crashing into something causes a continuous vibration until something else happens. I have only tested in GTAV as thats the only game I have that I use a controller with.

PS3 DS3 Controller in DS4 mode via BT. Controller model CECHZC2E
OS: Windows 10 20H2 19042.867// dshidmini_v1.0.50.0 // DS4windows 2.2.10 x64 // BthPS3 v1.3.127 // Vigem 1.17.333.0
Removed all traces of SCP and Shibari drivers using driver store explorer.

Is there some way I can produce a log for you with something useful in it?

nefarius, 2021-03-15:
Hello!

Yes, you're experiencing our latest "fix" to some controller locking up when they receive too many requests in a short amount of time. Therefore the latest version contains a first draft of a crude rate control logic, I can replicate what you experience as well so that's ofc. not ideal.

We're currently discussing proper solutions on the Discord, will update this issue if the Eureka moment struck 😉 

Cheers

nefarius, 2021-03-15:
> Same behaviour either wired or BT.

Oh, it's on USB as well?

ferociousdiablo, 2021-03-16:
> > Same behaviour either wired or BT.
> 
> Oh, it's on USB as well?

The inconsistent behaviour seems to happen on both. Like getting some vibration for events and then nothing for the same events seemingly random. However the function getting 'stuck' until the next command is sent only seems to happen on bluetooth. Apologies for my lack of clarity, I just did some further testing to confirm this.

A good test for the vibration getting stuck is the weapon wheel in GTAV, moving the stick around causes a short hard pulse which sometimes persists until you move it to another segment or after 5 seconds. Seems to happen after a little while of moving it around and is easily reproduceable.

It is very possible that these are 2 separate issues, but this is all well above my skillset to understand the code lol

nefarius, 2021-03-16:
We're still experimenting to find the sweet spot to make everyone happy 😉 Thanks for the feedback! 

Stay tuned!

nefarius, 2021-04-12:
Fixed in `v1.2.122`

closed 2021-04-12 by nefarius