nefarius/DsHidMini #19: Feature Request: Auto update in addition to update check for dshidmini

rjmxtechnology, 2021-03-17:
Currently I find myself manually copying the new exe over every time there is a new version etc... I think implementing something similar to scptoolkit's old update functionality where it did the work for you and replaced the old exe with the new one would be nice. Even something similar to ds4windows update functionality would be very helpful in my opinion.

nefarius, 2021-03-19:
I will do that once out of beta; I can not yet guarantee non-API-breaking changes between driver and UI so I'd need to also implement a strict driver version check and update that too.

nefarius, 2022-08-01:
News on this: since I have renewed my Advanced Installer subscription we can make an auto-updater a reality 🥳

nefarius, 2023-10-18:
Once we have a setup I'll use [vicius](https://github.com/nefarius/vicius) for the updating part as soon as it has matured.

nefarius, 2024-05-18:
No longer relevant.

nefarius, 2024-05-18:
Consolidated in #211.

closed 2024-05-18 by nefarius