nefarius/DsHidMini #27: controller disconnects 5-10s later if reconnected quickly after a disconnect

Kanuan, 2021-03-21:
Steps to reproduce (tested on bluetooth):

- Connect the controller as normal
- Disconnect/change mode and hit "apply"
- Quickly reconnect the controller
- Wait a few seconds and the controller will disconnect
- After trying to reconnect the controller again, even if done quickly, it will remain connected

morrowindmemes, 2021-03-21:
I can easily reproduce this too.

nefarius, 2021-03-21:
I can't reproduce, could you elaborate in more detail please 😅

Kanuan, 2021-03-21:
This video demonstrates the issue.

https://user-images.githubusercontent.com/24910442/111914579-4afcd600-8a51-11eb-8f1f-66579a713b60.mp4

The issue was tested on different computers, one with a USB BT dongle and the other being a notebook with built-in bluetooth card. On the video the controller was connected to the notebook, which was running the latest BthPS3 (v1.3.125) and DsHidMini driver (1.1.84v).
Also tested with 2 different controllers, same behaviour.

nefarius, 2021-03-21:
Oh yeah, I can reproduce it was well now 😅 How do you guys find these edge cases 😆 

The trace is clean so the driver ain't at fault:

```text
2021/03/21-18:35:48.570	 [DsHidMini_EvtDevicePrepareHardware][BTH] VID: 0x054C
2021/03/21-18:35:48.570	 [DsHidMini_EvtDevicePrepareHardware][BTH] PID: 0x0268
2021/03/21-18:35:48.570	 [DsHidMini_EvtDevicePrepareHardware] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:48.570	 [DMF_DsHidMini_Open] --> Entry
2021/03/21-18:35:48.570	 [DMF_DsHidMini_Open] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:48.570	 [DsHidMini_EvtDeviceD0Entry] --> Entry
2021/03/21-18:35:48.570	 [DsHidMini_EvtDeviceD0Entry] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:48.570	 [DsHidMini_EvtWdfDeviceSelfManagedIoInit] --> Entry
2021/03/21-18:35:48.570	 [DsBth_Ds3Init] --> Entry
2021/03/21-18:35:48.570	 [DsBth_Ds3Init] <--
2021/03/21-18:35:48.570	 [Ds_SendOutputReport] --> Entry
2021/03/21-18:35:48.570	 [DMF_OutputReportScheduledTaskCallback] --> Entry
2021/03/21-18:35:48.570	 [DMF_OutputReportScheduledTaskCallback]Time span since last packet was sent: 107421489 ms
2021/03/21-18:35:48.570	 [DsBth_SendHidControlWriteRequest] --> Entry
2021/03/21-18:35:48.570	 [DsBth_SendHidControlWriteRequest] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:48.570	 [DMF_OutputReportScheduledTaskCallback] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:48.570	 [Ds_SendOutputReport] <-- Exit <status=NTSTATUS=0x00000004>
2021/03/21-18:35:48.570	 [DsHidMini_EvtWdfDeviceSelfManagedIoInit]Disconnect event name: Global\DsHidMiniDisconnectEvent000704144951
2021/03/21-18:35:48.570	 [DsDevice_HotReloadConfiguration]Hot-reload triggered
2021/03/21-18:35:48.570	 [DsHidMini_EvtWdfDeviceSelfManagedIoInit]Configuration reload event name: Global\DsHidMiniConfigHotReloadEvent000704144951
2021/03/21-18:35:48.570	 [DsHidMini_EvtWdfDeviceSelfManagedIoInit] <-- Exit <status=NTSTATUS=0x00000004>
2021/03/21-18:35:48.597	 [DsBth_HidInterruptReadRequestCompletionRoutine]Battery status changed to 5
2021/03/21-18:35:48.597	 [DsBth_HidInterruptReadRequestCompletionRoutine]Updating battery status to 5
2021/03/21-18:35:48.597	 [Ds_SendOutputReport] --> Entry
2021/03/21-18:35:48.597	 [DMF_OutputReportScheduledTaskCallback] --> Entry
2021/03/21-18:35:48.597	 [DMF_OutputReportScheduledTaskCallback]Time span since last packet was sent: 26 ms
2021/03/21-18:35:48.597	 [DsBth_SendHidControlWriteRequest] --> Entry
2021/03/21-18:35:48.597	 [DsBth_SendHidControlWriteRequest] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:48.597	 [DMF_OutputReportScheduledTaskCallback] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:48.597	 [Ds_SendOutputReport] <-- Exit <status=NTSTATUS=0x00000004>
2021/03/21-18:35:48.670	 [DsBth_EvtControlReadTimerFunc] --> Entry
2021/03/21-18:35:49.585	 [DsBth_EvtControlWriteTimerFunc] --> Entry
2021/03/21-18:35:49.585	 [Ds_SendOutputReport] --> Entry
2021/03/21-18:35:49.585	 [DMF_OutputReportScheduledTaskCallback] --> Entry
2021/03/21-18:35:49.585	 [DMF_OutputReportScheduledTaskCallback]Time span since last packet was sent: 987 ms
2021/03/21-18:35:49.585	 [DsBth_SendHidControlWriteRequest] --> Entry
2021/03/21-18:35:49.585	 [DsBth_SendHidControlWriteRequest] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:49.585	 [DMF_OutputReportScheduledTaskCallback] <-- Exit <status=STATUS_SUCCESS (0x00000000)>
2021/03/21-18:35:49.585	 [Ds_SendOutputReport] <-- Exit <status=NTSTATUS=0x00000004>
2021/03/21-18:35:49.585	 [DsBth_EvtControlWriteTimerFunc] <--
2021/03/21-18:35:55.256	 [DsBth_HidInterruptReadRequestCompletionRoutine]Device has been disconnected
2021/03/21-18:35:55.258	 [DsBth_EvtControlReadTimerFunc] <--
2021/03/21-18:35:55.259	 [DsHidMini_EvtDeviceD0Exit] --> Entry
2021/03/21-18:35:55.259	 [DsHidMini_EvtDeviceD0Exit] <--
2021/03/21-18:35:55.259	 [DMF_DsHidMini_Close] --> Entry
2021/03/21-18:35:55.259	 [DMF_DsHidMini_Close] <--
```

I assume something within BthPS3 is the culprit but... While interesting and annoying.... Bottom lowest priority for me currently 😛

nefarius, 2022-08-01:
We should try to reproduce this again after the next BthPS3 release dropped, maybe it's fixed by the new PDO state machine 😇

morrowindmemes, 2022-08-02:
There isn't any beta participation to try this next release right?

nefarius, 2022-08-02:
> There isn't any beta participation to try this next release right?

Sure thing! On my Discord 😎

nefarius, 2022-08-04:
It's been confirmed that BthPS3 was at fault and is fixed in the upcoming release 🥳

closed 2022-08-04 by nefarius