nefarius/DsHidMini #34: DS3 => DS4 battery translation is incorrect

Kanuan, 2021-03-23:
DS4 battery info from the input report is translated as the following in ds4windows:

- 8-15 => 100 %
- 7 => 87 %
- 6 => 75 %
- 5 => 62 %
- 4 => 50 %
- 3 => 37 %
- 2 => 25 %
- 1 => 12 %
- 0 => 0 %

The current values used in the current version of dshidmini are:

- Full => 10 = 100%
- High => 8 = 100%
- Medium => 5 = 62 %
- Low => 2 = 25%
- Dying => 1 = 12%

I propose Full, High and Medium to be changed to 8, 6 and 4 to represent, respectively, 100%, 75% and 50%

nefarius, 2021-03-24:
Weird, I based my logic on the Linux sources who indicate, that 10 should equal 100%, 5 should be 50% etc.

Could @mika-n give some insights please? 🙂

mika-n, 2021-03-24:
@nefarius DS4Windows app actually treats the "max battery value from DS4" in a different way while the DS4 pad is connected to usb charger and the pad is charging and when the pad is in BT mode without charging anything. BATTERY_MAX_USB  vs BATTERY_MAX value. 
https://github.com/Ryochan7/DS4Windows/blob/jay/DS4Windows/DS4Library/DS4Device.cs#L1312

I don't know exactly is this logic scientifically proven to be correct or result of some test runs how genuine DS4 gamepad returns the raw battery value when the pad is fully charged or almost empty.

Anyway, then this raw battery value is re-mapped as 0-100% value using the MAX upper bound value.

nefarius, 2021-03-24:
Oh I see, that makes more sense now, thanks!

nefarius, 2021-04-12:
Fixed in `v1.2.122`

closed 2021-04-12 by nefarius