nefarius/DsHidMini #37: Right stick doesnt work properly

Symbolic1, 2021-03-24:
Hello people, the issue i'm facing appears when i try to configurate the buttons, the stick seems to be always pressed to only one direction and don't let me set the other buttons of the gamepad or the other directions of the same stick.
[image]

Kanuan, 2021-03-24:
not enough info to help.
What mode the controller is currently on?
Send screenshots from the Windows' "game controllers" (search on the start menu). Like this:
[image]

Also, if your intention is Xbox or Dualshock 4 emulation, why not going into the [DS4Windows route](https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/)?

Symbolic1, 2021-03-24:
What do you mean with "mode"?

DS4 doesn't worked for me, i used to connect through Shibari, but it didn't worked anymore since a couple of days. I followed all the steps to delete the Shibari stuff for use DsHidMini

[image]

Kanuan, 2021-03-24:
It's all in the documentation, and you should read it. We can't offer more help for things that are already covered

[General features of DsHidMini](https://github.com/ViGEm/DsHidMini#features)

[HID Device Modes Explained](HID Device Modes Explained)

[DS4 Mode Guide](https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/)

Symbolic1, 2021-03-24:
Ok, thank you for your help

Kanuan, 2021-03-24:
If you managed to fix your issue, let us know. If you still have questions or need one or other sugestion you can ask more directly on the Discord server in the support channel, but please, read first the documentation that was linked above.

If you intended to use your PS3 controller as a Xbox controller (__Xinput emulation__) the easier route is by following the DS4 Mode Guide that was linked

nefarius, 2021-03-25:
Closed since the docs cover the configuration and the recommended support channels.

closed 2021-03-25 by nefarius