nefarius/DsHidMini #39: Input was inverted (Triangle became Cross)

Benja-S, 2021-03-24:
Hello! After updating to DsHidMini, from scptoolkit on one PC (removed all drivers), I realized the input became weird: 
Cross became Triangle (Y became A). 

I updated my main PC from Shibari + Fireshock (also removed Fireshock drivers as per instructions), to DsHidMini and found the exact same issue, I tried with 2 original DS3 controllers, and one offbrand (fake) with the same results. Also, on my main PC (desktop), DsHidMini will not work via bluetooth (with the USB Dongle that worked perfectly with Shibari + Fireshock).

Kanuan, 2021-03-25:
This message was copy-pasted from elsewhere, so it _may_ not suit exactly your situation, but read it to the end nonetheless. 
------------------------------
If your intention is to use your Dualshock 3 controller as a Xbox controller or as a Dualshock 4, keep something in mind: you need some _other_ program to translate the Dualshock 3 into a xbox / DS4 controller. Also, "Xbox controller" emulation is more commonly called __XInput__ emulation. Got it?

Whatever you used before to connect your DS3 controller into the PC ( ScpToolKit / Shibari / Better3DS / MotionInJoy ) presented your DS3 as a Xbox controller by default. This is __not__ the case with DsHidMini, as it presents the controller as a generic desktop controller (__DInput__). There are reasons for this that I'll not touch here but you can ask in the Discord server linked somewhere below if you are really interested.

If you just want to use your DS3 as a Xbox controller (Remember, __XInput__) or as a Dualshock 4, the easier route is by following the [DS4 Mode User Guide](https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/). Though the name is "DS4Windows", it will emulate XInput by default when it picks the controller, already mapped, with no need of more configuration in this case. Actual DS4 emulation with DS4Windows needs to be selected inside the profile settings => other tab.

If you want to use _another_ program that is not DS4Windows for your XInput / DS4 emulation, then you need to know about the [HID Device Modes](https://vigem.org/projects/DsHidMini/HID-Device-Modes-Explained/) your controller can be set into and their differences. Actually, you should know about them anyway, so give it a good read. Keep in mind that __no mode will emulate XInput by itself__ as I already pointed out somewhere above, so don't go changing between them and wondering why it's not working as a xbox controller.

For other features (quick disconnection combo, automatic disconnect when idle), you should read the Features section of the [DsHidMini's github main page](https://github.com/ViGEm/DsHidMini#features).

Now, if you have a question that is ___not___ covered in one of the pages linked above, feel free to ask in the [Discord server's support channel](https://github.com/ViGEm).

Kanuan, 2021-03-25:
Regarding the bluetooth issue, unless you discovered some unknown DsHidMini bug (not impossible, just unlikely) it should 100% work since it worked with Shibari. Are you sure your BthPS3 is updated? Unninstall it from windows first by using the built-in uninstaller then install the [latest BthPS3 version](https://github.com/ViGEm/BthPS3/releases).

Benja-S, 2021-03-25:
> Regarding the bluetooth issue, unless you discovered some unknown DsHidMini bug (not impossible, just unlikely) it should 100% work since it worked with Shibari. Are you sure your BthPS3 is updated? [...]

ok, so I'm juggling with DS4 hide thing now to fix the dual input problem, all seems to be going good so far, however, I did update BthPS3 as instructed (uninstalling beforehand) but still DsHidMini will not pick up my controller via bluetooth (the dongle is working correctly, as I am able to connect other devices via bluetooth), with SCP and Shibari I was able to see if my dongle was being picked up, but with DsHidMini there is no such info, so all I know is bluetooth does not seem to be working (with my notebook it's working seamlessly, without the need for an external dongle, even)

Benja-S, 2021-03-25:
> ok, so I'm juggling with DS4 hide thing now to fix the dual input problem, all seems to be going good so far, however, I did update BthPS3 as instructed (uninstalling beforehand) but still DsHidMini [...]

Ok, so I checked BthPS3 driver config tool and the bluetooth radio (dongle) is being picked up just fine, also DsHidMini shows BthPS3 options so the conflict doesn't seem to lie there. So still, I'm 100% sure the dongle is not at fault here and yet the controller will not pair with this new software.

Benja-S, 2021-03-25:
> Ok, so I checked BthPS3 driver config tool and the bluetooth radio (dongle) is being picked up just fine, also DsHidMini shows BthPS3 options so the conflict doesn't seem to lie there. So still, I'm [...]

Ok sorry for so many self-replies but after about 15 tries (disconnecting, waiting and reconnecting) the controller has paired and connected succesfully. It also seems, however, that after a few seconds of inactivity the controller did disconect. It took a few tries (only about 3 this time) and It has connected again, this time with no disconnection so far. So everything seems to be working as intended for now, thank you for the help!

Kanuan, 2021-03-25:
Nice to know that everything is working as normal. 

Be warned that there is a bug right now affecting some specific controller models that ___may___ cause the controller to _miss_ the "stop rumbling" command on bluetooth, making it vibrate for more than than it should or even until a new rumble command is received. However, this issue is not a reason to stop using DsHidMini, since it probably plagues every single DS3 program/driver since forever.  [We now know what causes the issue](https://github.com/ViGEm/DsHidMini/issues/20) and Nefarius is planning on fix this in the up-coming weeks. 

Besides that, enjoy!

closed 2021-03-25 by nefarius