nefarius/DsHidMini #40: Read this if your intention is using your PS3 controller as a PS4 or Xbox Controller

Kanuan, 2021-03-26:
First, know the terms:

- Standard PS3 controller => DualShock 3 (__DS3__)
- Standard PS4 controller emulation => DualShock 4 (__DS4__) emulation
- Xbox controller emulation => __XInput__ emulation

If your intention is to use your DS3 as a Xbox or DualShock 4 controller, keep something in mind: you need some _other_ program alongside DsHidMini to emulate a __XInput__ or a DS4 device.

Whatever you used before to connect your DS3 into the PC ( ScpToolKit / Shibari / Better3DS / MotionInJoy ) presented your DS3 as a Xbox controller by default. This is __not__ the case with DsHidMini, as it presents the controller as a generic desktop controller (__DInput__ device). There are reasons for why it works this way that I'll not touch here, but you can ask in the Discord server linked somewhere below if you are really interested.

If you just want to use your DS3 as a Xbox controller (Remember, __XInput__ device) or as a Dualshock 4, the easiest route is by following the [DS4 Mode User Guide](https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/). Though the name of the program used in the guide is "DS4Windows", it will emulate XInput by default when it picks the controller, already mapped, with no need of further tweaking the configuration in this case. If you want DS4Windows to emulate a DS4 instead you can do so by selecting it in the "profile" settings => other tab. Read the _DS4 Mode User Guide_ to the end to not run into issues, everything is explained there.

If you want to use _another_ program that is not DS4Windows for your XInput / DS4 emulation, then you need to know about the [HID Device Modes](https://vigem.org/projects/DsHidMini/HID-Device-Modes-Explained/) your controller can be set into and their differences. Actually, you should know about them anyway, so give it a good read. Keep in mind that __no mode will emulate XInput nor DS4 by itself__ as I already pointed out somewhere above, so don't go changing between them and wondering why it's not working as you want to. 
__(If you plan on using other modes/programs for your XInput / DS4 emulation, be sure to read the post below this one)__

For other features (quick disconnection combo, automatic disconnect when idle), you should read the Features section of the [DsHidMini's github main page](https://github.com/ViGEm/DsHidMini#features).

Now, if you have a question that is ___not___ covered in one of the pages linked above or just want to say hi, feel free to join the [Discord server](https://github.com/ViGEm).

Kanuan, 2021-03-26:
TO DO: SECTION ABOUT USING OTHER MODES FOR XINPUT / DS4 EMULATION AND EXPLAIN THE DOUBLE INPUT ISSUE THAT MAY ARISE
--------

jdgp31, 2021-03-31:
Thank you for this pinned post !  It may help others as it did me. Wasn't aware when setting up DsHidMini for my PS3 controller that I needed another program (DS4Windows) to emulate an Xbox 360 controller in game. Working perfectly now,

OneCricketeer, 2021-06-20:
Anecdotally, my DS3 Controller worked as expected (via Stream controller support) for several hours after uninstalling SCPToolkit and installing this 

Only after unplugging and rebooting did I start experiencing issues in my game where it seems to think the bumper triggers are inverted/permanently pressed down. Windows USB Controller dialog says differently, however. Would also be nice if X and O buttons were actual buttons and not seen as sliders

nefarius, 2021-06-20:
> Anecdotally, my DS3 Controller worked as expected (via Stream controller support) for several hours after uninstalling SCPToolkit and installing this
>
> Only after unplugging and rebooting did I start [...]

Please don't hijack threads and open a Discussion in case there's a reproducible demand for support, thanks.

nefarius, 2022-08-01:
Closing to clean the issue tracker but leaving it pinned.

closed 2022-08-01 by nefarius

nefarius, 2024-05-17:
@Kanuan can you please edit the first post and update the now dead links, thanks.