nefarius/DsHidMini #41: DS3 shoulder triggers inverted

jpof88, 2021-03-27:
DS3 shoulder triggers are inverted to me. 100% when not pressing and 0 when fully pressed.
I installed BthPS3 and dshidmini_v1.1.84.0

Anything I'm doing wrong? 

Thank you.

nefarius, 2021-03-27:
What mode? What game? Have you read the documentation?

jpof88, 2021-03-27:
Hey,

Thanks for the quick reply.

I see what you mean...It was on SXS RPCS3 mode. Changed to GPJ Generic Direct input and it is now fine after calibration.
I am using it over bluetooth. is this a good setup or should I try something different?

Thank you.

jpof88, 2021-03-27:
quick update. DS4 mode seems to be the best for me.

nefarius, 2021-03-27:
> I see what you mean...It was on SXS RPCS3 mode. Changed to GPJ Generic Direct input and it is now fine after calibration.
> I am using it over bluetooth. is this a good setup or should I try something different?

That's the beauty of this design; **you** configure it the way **you** get best value out of it 😁 

> quick update. DS4 mode seems to be the best for me.

There ya go 👍

jpof88, 2021-03-27:
I used scp toolkit for years but switching to this now.

Great job. Very appreciated

closed 2021-03-27 by nefarius