nefarius/DsHidMini #43: Steam (Specifically rocket league) will not see the controller as DS4

KueDAWG, 2021-04-02:
Hi there Nef!   Amazing work as always.   Just thought I'd bring this up as an issue (May not be.  not sure).   So with shibari/Fireshock/BTHPS3 combo steam saw the ps3 controller as a PS4 and so did rocketleague.  Worked flawlessly.   However DSHidMINI seems to struggle.  any of the 4 modes allow steam to see the control (Can maneuver big pic mode no prob) but rocketleague will not recognize any of the inputs.  I tried using the DS4 guide but that did not work either.  The only way I can get RKL to see it is to use SXAXIS mode on dshidmini and on steam enable Playstation controller support.   This allows input but while steam sees it as a playstation control, rkl sees it as xbox.  weird!    In any case, this may not qualify as a bug but since it worked on the old setup and not on this one, I thought it might.

nefarius, 2021-04-02:
It heavily depends how you set it up because Steam tries a lot of compatibility fixes for games in a transparent fashion. I'd have suggested you use SXS and enable PlayStation Controller support in Steam and be good to go. Steam presents it as an Xbox controller since the game has no native support for the DS3, it does the convenience work for you.

nefarius, 2021-04-02:
It's not a bug, you need to find the best configuration for your specific case yourself I'm afraid.

KueDAWG, 2021-04-02:
Gotcha!   Kewness.  sxs definitely worked, but I think I'll continue to use the old setup for now, as I like the PS buttons to show up in rocket league's in game menus, etc.  Out of curiosity, will there be any plans to make the PS3 control show up as DS4 using dshidmini the same way it showed up as DS4 with fireshock and Shibari?  I read and used the DS4 guide instructions but it still shows up on steam as an xbox controller.  Obviously not a necessary feature.  just curious if it was planned for future.

nefarius, 2021-04-02:
You can already do exactly that. Simply follow the DS4 guide and emulate a DS4 instead of X360 and it does exactly the same as the Shibari setup did.

Kanuan, 2021-04-02:
https://github.com/ViGEm/DsHidMini/issues/40

nefarius, 2021-04-03:
I'll allow myself to close since everything is working as intended and documented 👻

closed 2021-04-03 by nefarius