nefarius/DsHidMini #45: Still says that there is a newer version available

johnodon, 2021-04-02:
[image]

nefarius, 2021-04-02:
When did you update to this version? Today?

johnodon, 2021-04-02:
> When did you update to this version? Today?

Yes sir.

nefarius, 2021-04-02:
Ah that is a bug of my ingenious caching routine 🐛 Please close it, go to `%APPDATA%` in File Explorer and delete the DsHidMini....json file and try again.

johnodon, 2021-04-02:
That did the trick.  TY

nefarius, 2021-04-02:
Good to know, I'll remove that "feature" in the next release as it clearly does more harm than help.

Kanuan, 2021-04-02:
Maybe replace it with a "check if there is a update available" works better?

nefarius, 2021-04-04:
> Maybe replacing it with a "check if there is a update available" buttons would be better?

Yep, I think that's even better. Gives the user control over when to contact the GitHub API and speeds up application start.

nefarius, 2021-04-06:
Reopened to stay visible until next official release is published.

nefarius, 2021-04-12:
Fixed in `v1.2.122`

commit 9cb6bb3c (2021-04-04): Fixed #45
closed 2021-04-12 by nefarius