nefarius/DsHidMini #48: Add support for Navigation Controller

nefarius, 2021-04-06:
## Open tasks

- Add hardware IDs back into INF
- Adjust LED behaviour
  - Only one LED, so make slow flash during charging, solid on when full and rapid flashing when low?
- Check code for possible regressions
  - Should be fine overall since the Navigation Controller responds to the same feature requests and has the same input report length and format, it simply reports less buttons/axes
- Some modes will not behave as expected
  - DS4Windows LED translation will not really do much

expected-ingot, 2022-08-10:
when?

nefarius, 2022-08-10:
IDK

JSRRosenbaum, 2022-12-20:
@expected-ingot 
I was able to get the Navigation controller to work by copying the hardware ID's back into the INF, as you had mentioned. 

I added the following at the end of the line with the comment : "NAVIGATION wireless via BthPS3"
&Dev&VID_054C&PID_042F

I also had to add the following lines after the original XINPUT adapter entries: 

%dshidmini.GIP.HID.DeviceDesc%=dshidmini.GIP.HID, HID\VID_054C&PID_042F&MOD_05
%dshidmini.GIP.HID.DeviceDesc%=dshidmini.GIP.HID, HID\{206f84fc-1615-4d9f-954d-21f5a5d388c5}&Dev&VID_054C&PID_042F&MOD_05

I was able to add the modified driver using the script located here: 

https://github.com/fawazahmed0/windows-unsigned-driver-installer

I'm happy to report everything seems to be working fine with no issues discernable yet. 

I've included a zip below containing the modified inf's and the installer (use at your own risk!) 

[dshidmini.zip](https://github.com/ViGEm/DsHidMini/files/10271331/dshidmini.zip)

nefarius, 2026-09-08:
Implementation for Navigation Controller support is on `feat/48-navigation-controller`.

**Done in software**
- Device type is classified from `VID_054C` / `PID_042F` and published as `DEVPKEY_DsHidMini_RO_DeviceType` (pid 12).
- One-LED battery indication: rapid flash when low/dying, slow flash while charging, solid when charged, off when unknown. Custom/application LED output is clamped to LED 1. The four-LED USB charging chase does not run.
- Rumble duration/strength bytes are cleared at the output-report send boundary. ControlApp hides rumble groups and extra LED editors.
- USB `USB\VID_054C&PID_042F` and BthPS3 `BTHPS3BUS\{206F84FC-1615-4D9F-954D-21F5A5D388C5}&Dev&VID_054C&PID_042F` are restored in both Win10 and Win11 INF sections.
- Notes: `docs/NAVIGATION_CONTROLLER.md`. Checklist: `research/navigation-controller-matrix.md`.
- `.\build.cmd Compile` succeeded. ControlApp.Tests: 123 passed.

**Still required before close**
Fill the USB and Bluetooth hardware matrix on a genuine CECH-ZCS1. Do not close this issue until both columns pass.

**Limits**
XInput is recommended. SDF/GPJ/SXS/DS4Windows remain available for partial-input compatibility. Missing buttons, the right stick, extra LEDs, and rumble stay absent or have no effect. Pairing still follows the #321 soft-fail path if `0xF2`/`0xF5` are unsupported.

nefarius, 2026-11-12:
Status update (2026-11-12), based on inspection of `master` at [`1bc05e1`](https://github.com/nefarius/DsHidMini/commit/1bc05e1a7fd2f43318f094ca18ed76f09d37f511).

**Implementation is merged**

The implementation previously reported on `feat/48-navigation-controller` is now in `master` via #501:

- USB and BthPS3 hardware IDs for `VID_054C` / `PID_042F` are present in the Windows 10 and 11 INF sections.
- Navigation is identified as one-LED, no-rumble hardware. Battery/custom LED output uses LED 1, the four-LED charging chase is suppressed, and motor duration/strength bytes are cleared at the output send boundary.
- ControlApp adapts its controls to those capabilities.

Follow-ups #506 and #507 fix Navigation status reporting and stop restarting the charging LED flash.

**Documentation is present**

[docs/NAVIGATION_CONTROLLER.md](https://github.com/nefarius/DsHidMini/blob/1bc05e1a7fd2f43318f094ca18ed76f09d37f511/docs/NAVIGATION_CONTROLLER.md) covers available controls, mode limitations, LED meanings, and pairing. XInput is recommended; other modes expose a partial controller and cannot supply missing controls or rumble. The issue's unchecked documentation item appears to lag the repository state.

**Still required before closing**

The [hardware acceptance matrix](https://github.com/nefarius/DsHidMini/blob/1bc05e1a7fd2f43318f094ca18ed76f09d37f511/research/navigation-controller-matrix.md) still has no recorded USB or Bluetooth results. Complete it on a genuine CECH-ZCS1, especially:

- Device MAC read, host MAC read/automatic pairing, custom pairing address, and wireless reconnect after USB pairing.
- Present inputs and neutral absent controls, one-LED battery/charging behavior, and no rumble side effects.
- HID mode checks, disconnect/hot reload, and sleep/resume on both transports.

Pairing continues to use the DS3 `0xF2` / `0xF5` path, with the #321 soft-fail behavior if unsupported; its presence in code is not evidence that it works on Navigation hardware.

This update is a source/history inspection, with no fresh build, automated test run, or live controller test. The earlier successful build and 123-test result are historical software checks. The remaining blocker is documented hardware acceptance on both USB and Bluetooth; keep the issue open until those results are recorded. Merged in `master` does not establish availability in a published release.