nefarius/DsHidMini #50: Decode and utilise potential device identification report packet

nefarius, 2021-04-06:
Related to #16, #20 and possibly #3

[Some interesting unused packets exchanged by controller and PS3](http://eleccelerator.com/wiki/index.php?title=DualShock_3#Other_Initial_Control_Transfers) have been documented for ages but not really further decoded and documented.

## Example implementation

As documented on the Wiki send a Class Interface GET_REPORT request with value `0x0301` to the controller which returns approximately 44 bytes of (non-zero) data.

```c
UCHAR tmpBuf[64];

if (NT_SUCCESS(USB_SendControlRequest(
	pDevCtx,
	BmRequestDeviceToHost,
	BmRequestClass,
	GetReport,
	0x0301, // "Identification" request(?)
	0,
	tmpBuf,
	64
)))
{
	DumpAsHex("EXPERIMENTAL", tmpBuf, 64);
}
```

## Packet dumps

After a bit of testing with various controllers I have in my stash I am delighted to observe a consistent pattern of responses with different (yet undisclosed) fields per controller model/revision 🙂 

```text
Eleccelerator Wiki	
00010400070C010218181818090A1011121300000000040002020202000000040404040000040001020700170000000000000000000000000000000000000000

CECHZC2E A1
00010400080C010218181818090A1011121300000000040002020202000000040404040000040001020700170000000000000000000000000000000000000000

CECHZC2U A2
000104000B0C010218181818090A1011121300000000040002020202000000040404040000040001020700170000000000000000000000000000000000000000

CECHZC2E B1
00010401030C010218181818090A1011121300000000040002020202000000040404040000030001020000170000000000000000000000000000000000000000

SIXAXIS
000102000308010217171717090A0000000000000000040002020202000000040404040000010600000000000000000000000000000000000000000000000000

PANHAI
00010300050C010218181818090A1011121300000000040002020202000000040404040000020102006400170000000000000000000000000000000000000000
```

More to come 😁

nefarius, 2021-04-06:
Exposed the packet data as property for easier access to the user (no tracing required):

[image]

Kanuan, 2021-04-06:
`Model: CECHZC2M (Date Code: 2D) / Dualshock 3
00 01 04 00 08 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 04 00 01 02 07 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

Model: CECHZC2M (Date Code: 3A) / Dualshock 3
00 01 04 00 09 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 04 00 01 02 07 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

Model: CECHZC1U / FCC ID: AK8CECHZC01 / IC: 409B-CECHZC01 / Sixaxis
00 01 03 00 05 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 01 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
`

Hubert365, 2021-04-07:
### Model: CECHZC2U / FCC ID:AK8CECHZC2 / IC: 409B-CECHZC2

`00 01 04 00 06 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 04 00 01 02 07 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 `

nefarius, 2021-04-12:
This property is included since release `v1.2.122` (**USB only**). Use Device Manager to read.

nefarius, 2021-05-02:
Reported by Abi on Discord (fake, does not work, Windows BTHUSB rejects it because no channel encryption):

```text
00 01 03 00 05 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 
00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 02 01 02 
00 64 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 
```

nefarius, 2024-08-12:
Reported from Mic on Discord.

### CECHZC2U with working Gyro

```
00 01 04 00 07 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 
00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 03 00 01 
02 00 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00
```

### CECHZC2U with Gyro requiring calibration

```
00 01 04 00 07 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 
00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 04 00 01 
02 07 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00
```

nefarius, 2026-09-24:
We did the best we could with this info, closing.

closed 2026-09-24 by nefarius