nefarius/DsHidMini #52: Big Motor only rumbles above certain strength value

Kanuan, 2021-04-07:
### Summary

Big Motor (BM) commands sent to the controller have two parameters: duration and strength. The values range from decimal 0 to 255. Regarding rumble strength:

- decimal 0 turns off the BM
- decimal 1 _should_ activate the BM at minimum strength
- decimal 255 activates BM at maximum strength

What actually happens is that the BM is only activated at a value far greater than decimal 1. With a controller Model: CECHZC2M (Date Code: 2D) the minimum value is decimal 64.

### Testing

- If possible, do these tests with the controller connected by bluetooth
- [Set the controller in SXS mode](https://vigem.org/projects/DsHidMini/HID-Device-Modes-Explained/)
- Make sure the controller is not hidden by HidGuardian/HidHide. If the controller is hidden with HidHide you can just open its Configuration Utility, disable the "Enable device hiding" box in the devices tab and then reconnect the controller to make it visible again.
- [Download and extract this archive](https://github.com/ViGEm/DsHidMini/files/6272517/RumbleStrengthTest.zip)
- Run "MightyHIDTest.exe". If it immediately closes then your controller is not in SXS mode or is hidden
- __MAKE SURE TO NOT CLICK INSIDE THE BLACK WINDOW OTHERWISE THE PROGRAM WILL CRASH__. If you need to select the window to write make sure to click on the top of the command window
- Run the test once (write "1" and then ENTER). Your controller should rumble weakly. If it does not, then the minimum value that makes your controller rumble is greater than decimal 64
- Close the command window

Now, to discover what is the minimum strength that makes your controller rumble, we need to edit the `settings.json` file in the extracted folder to vary the strength value. First, let's choose a new strength value. 

If your controller rumbled with decimal 64, then the next lower value is decimal 63. If it didn't rumble then let's go higher with decimal 65

The reason I'm writing "decimal" everywhere is because we will need to use values in hexadecimal. E.g.:  

- decimal 65 is hexadecimal 41
- decimal 64 is hexadecimal 40
- decimal 63 is hexadecimal 3F

So, after choosing the new decimal value you want to test you need to find its hexadecimal representation. You can convert from decimal to hexadecimal by using [this site](https://www.rapidtables.com/convert/number/decimal-to-hex.html) or by looking in the table in the image below, where hexadecimal values are in black and decimal in red:
[image] 

After having chosen the hexadecimal representation of the new test value:
- Open the `settings.json` file with a text editor (windows' notepad is enough, just right click on the `settings.json` file -> Open with -> Choose another app -> Select notepad on the list
- REPLACE the first 2 digits ___AFTER_ THE "7F" AS SHOWN IN THE IMAGE BELOW__ with the new value:
[image]
- Close the notepad and choose to save changes
- Run MightyHIDTest.exe again, choose "1" again
- Check if your controller will rumble
- Keep trying different values until you find the minimum value where the controller rumbles

### Reporting your findings

It would be great if you provided the maximum amount of info regarding your controller and the tests:

- The minimum value that makes your controller rumble (specifiy if the value is in Decimal or hexadecimal. Better yet, inform both)
- A print of your DSHMC tool so it's possible to see the genuine controller check
- [The identification property of your controller](https://github.com/ViGEm/DsHidMini/issues/50). Keep in mind that 1) you need to be on the latest released BETA driver 2) be connected by USB for this info to be available on Windows. [REMIND HERE: FIX LINK AND INFO LATER]

### Proposed solution:

Re-scale BM Strength values to the range that actually activates the motor. If the minimum value is not consistent across all devices then include in DSHMC tool a  slider to adjust the minimum value used for the re-scaling function.

Kanuan, 2021-04-07:
# This is just an example of report. Don't worry about organizing everything perfectly

## Model: CECHZC2M (Date Code: 2D) / Dualshock 3

__Minimum strength value that activates rumble__: decimal 64 = hexadecimal 40

__Identification property on windows (can only be check when connected by USB)__:

```text
00 01 04 00 08 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 04 00 01 02 07 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```

### __Genuine?__: Yes

[image]

## Model: CECHZC2M (Date Code: 3A) / Dualshock 3

__Minimum strength value that activates rumble__: decimal 64 = hexadecimal 40

__Identification property on windows (can only be check when connected by USB)__:

```text
00 01 04 00 09 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 04 00 01 02 07 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```

__Genuine?__: Yes

[image]

Hubert365, 2021-04-07:
[image]

### Minimum strength value that activates rumble: _`decimal 64 = hexadecimal 40`_

### Model: CECHZC2U / FCC ID:AK8CECHZC2 / IC: 409B-CECHZC2

```text
00 01 04 00 06 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 04 00 01 02 07 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```

Ghaleon4, 2021-04-09:
## Model: CECHZC2U (Date Code: ?) / Dualshock 3
__Minimum strength value that activates rumble__: decimal 64 = hexadecimal 40

Identification property on windows (can only be check when connected by USB):
Not queried yet 

__Genuine?__: NO
[image]
[image]

nefarius, 2021-04-12:
FYI release `v1.2.122` has the identification packet feature included.

Kanuan, 2021-05-11:
## CONTROLLER SPECS
Model: CECHZC2U / Dualshock 3
FCC ID: AK8CECHZC2
IC: 409B-CECHZC2

__Minimum strength value that activates rumble:__  couldn't find the exact value, but it is around decimal 84 = hexadecimal 54

__Identification property on windows (can only be check when connected by USB):__
```text
00 01 04 00 06 0C 01 02 18 18 18 18 09 0A 10 11 12 13 00 00 00 00 04 00 02 02 02 02 00 00 00 04 04 04 04 00 00 04 00 01 02 07 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

__Genuine?:__ Yes
[image]

## Notes

Because this controller's minimum BM rumble strength isn't 0x40, it seems like the option to tweak the minimum value is necessary.

PR #219 (2022-04-26): Introduces new JSON-based configuration system
## What's changed

- Fixes #196 
- Fixes #81 
- Fixes #53 
- Fixes #52 
- Fixes #51 
- Fixes #35 
- Fixes #30 

## What's missing

- DSHMC needs a total overhaul as this change is 99% incompatible with the current config system
- See open tasks in #197 for open work
PR #218 (2022-05-06): Improved handling of force feedback/rumble
# What's changed

- Implements #51 
- Fixes #52
- Now will allow disabling the big and small motors individually
- Implemented  some logic in the configuration loading function to verify if valid values are being used for the rumble rescaling related functions
    - If invalid values are found (MinValue= 0, or MinValue >= MaxValue) then the corresponding rescaling function is disabled along with other associated features
- Fixed ForcedSM BM/SM threashold value not being properly loaded (see 48ac1790036c955f10e6cd16b9838d6beb86ee71 )

# What's missing
 [... cut]
closed 2022-04-26 by nefarius