nefarius/DsHidMini #55: Connecting over Bluetooth?

YourKalamity, 2021-04-08:
I have BthPS3 installed from my previous Shibari.DOM.Server installation but the SIXAXIS controller I use does not automatically reconnect when disconnected from USB and pressing the PlayStation button.
How can I pair the controller to work over Bluetooth?

Kanuan, 2021-04-08:
If you were using the controller over Bluetooth with Shibari then the controller _probably_ already is paired to the radio. Assuming you are on the latest BETA of DsHidMini, you can try:

- Making sure that DsHidMini is picking the controller by checking if it appears on the [DSHMC tool](https://vigem.org/projects/DsHidMini/HID-Device-Modes-Explained/)
- Connecting your controller by USB and [verifying in the DSHMC tool if the controller is paired to the current host](https://github.com/ViGEm/DsHidMini/issues/10)
- Checking if you are on the latest BthPS3. If you are not sure unninstall it by the control panel and install the latest version
- Reseting the controller by pressing/holding the Reset button in the back of the controller then connecting by USB again for DsHidMini

YourKalamity, 2021-04-08:
Reinstalling BthPS3 fixed this, thanks for the support :)

closed 2021-04-08 by YourKalamity