nefarius/DsHidMini #60: Dualshock 3 controller does not work

raul3d, 2021-04-11:
Sony Dualshock 3 controller connected to the PC with the default USB cable. Tried connecting it to a front USB 2.0/3.0 or directly to an motherboard USB 2.0 connector. DSHMC shows the "No compatible devices found" message. No previous DS3 drivers were installed.

Windows 10 Pro 20H2 64bit
DsHidMini 1.1.84.0 (x64)

Windows device manager shows the controller with device type "Nefarius HID Devices". The device status is (german Windows, I just post the orginal messages):
Das Gerät kann nicht gestartet werden. (Code 10)

{Gerät-Zeitüberschreitung}
Das Zeitlimit des angegebenen E/A-Vorgangs auf %hs wurde erreicht, bevor der E/A-Vorgang abgeschlossen wurde.

The windows event log shows this:
Beim Start des Geräts USB\VID_054C&PID_0268\5&887acb0&0&6 ist ein Problem aufgetreten.

Treibername: oem17.inf
Klassen-GUID: {2b959c9d-a4ed-4464-809d-0f1118cf055d}
Dienst: mshidumdf
Untere Filter: WUDFRd
Obere Filter: 
Problem: 0xA
Problemstatus: 0xC00000B5

What kind of information could help you narrow this down?

Kanuan, 2021-04-11:
Try resetting your controller with the reset button on its back

raul3d, 2021-04-11:
Thanks. That did the trick. The controller is now correctly detected and shows up fine in DSHMC.

Do you have an idea why the reset was necessary? What caused the issue? I was using this controller fine with a previous Windows 8.1 installation without even knowing about the reset button. Might help others if it is something more widespread.

Kanuan, 2021-04-11:
I don't know the details, just that it works. I'll add this part to the "how to install" troubleshooting part

nefarius, 2021-04-11:
> Thanks. That did the trick. The controller is now correctly detected and shows up fine in DSHMC.
>
> Do you have an idea why the reset was necessary? What caused the issue? I was using this controller [...]

Sometimes the "firmware" gets locked up, even with a hardware PS3, the reset button clears any saved settings it has and 99.9% fixes all of these weird issues. It is actually wide spread knowledge, just among enthusiasts, not the common PC user 😉

closed 2021-04-12 by nefarius