nefarius/DsHidMini #61: Fake/Copy-cat DS3 controller immediately disconnects when connected by bluetooth

Kanuan, 2021-04-11:
# The issue

Some reeeally rare copy-cats controllers don't use encrypted connections on bluetooth. Because Windows sees this unsecure connections as a secure hazard it immediately drops the device when it tries to connect. This is something that DsHidMini and BthPS3 can't fix, the only option is to use the controller wired.

# Know controllers with this issue

There's literally only one known controller, some Amazon DS3 copy-cat identified as _Pro Controller_.

# How to check

Follow the steps here on troubleshooting everyting to be 100% sure DsHidMin is correctly installed, the controller is correctly paired to the current Bluetooth Host address and everything indicates the controller _should_ be connecting.
After the device tries to connect and Windows drops it a warning is created on Windows' event log. To check, go in -> ??? [To-do].

[image]

# Working around the issue

The only way is fully replacing the whole Windows' Bluetooth Stack with another KMDF bluetooth stack that checks and accepts in-coming, unsecure DS3 connections. __This driver does not exist__ and probably never will since it's a enormous amount of work for little to know benefit (how many people use this _Pro Controller_ DS3 copy-cat?).

nefarius, 2021-04-11:
[Related important read](https://vigem.org/projects/BthPS3/Amazon-Choice-Pro-Controller/).

Kanuan, 2021-05-15:
# Known unsupported controllers

## PS III Double Shock Wireless Controller

Besides the amazon one Nef linked above, this `DUALSHOCH` controller from... Wantech Electronics? I think, seems to also use unsecured connections via BT, meaning it also is unsupported by Windows/BthPS3

[image]
[image]

Akami23, 2021-07-27:
i have a same ds3 controller name pro controller, its connected to laptop wirelessly but its disconnected after 6 seconds.
What is the problem? in the controller or in the dongle?
[image]

nefarius, 2022-08-23:
Closed so it disappears from the open issues list but keeping it pinned.

closed 2022-08-23 by nefarius