nefarius/DsHidMini #68: SDF mode breaks compatibility with older games

Kanuan, 2021-04-15:
# The issue

SDF mode sliders number go beyond the DInput limit. This seems to be breaking compatibility with some older games (e.g.: Legacy of Kain: Defiance). When setting the gamepad in-game some buttons don't register and the ones that do are out-of-order. In GPJ mode the buttons are mapped correctly.

# To-do

Remove all sliders that reside outside of DInput axis limit and test if this fixes game compatibility to verify if this really is what is causing the issue.

# Proposed solution

A new HID Device Mode (Common Gamepad) that spawns a similar layout to SDF but without the buttons' pressure sliders. This mode would be the ideal mode for  old games compatibility since it spawns only one device and by not having additional sliders there's no chance of the pressure sliders messing when setting button/axis mappings.

Kanuan, 2021-04-16:
# Hypothesis

The number of total axis in SDF may break compatibility with some games.
SDF Mode layout has:

- 13 buttons
- 1 Axis for the POV T-hat
-  6 axis for both sticks and triggers
- 10 Sliders for the pressure feature in each button (d-pad, face and R1/L1)

Because of the limitations of D-Input, 8 of the sliders are normally out of reach.

# Tests

First, it was tested if the SDF layout was broken beforehand when the [unsued 14th button was removed](https://github.com/ViGEm/DsHidMini/pull/26), but after reverting to a version of DsHidMini before the change it was comproved the issue still occurs.

Second, it was tested if the number of total axis in the controller was breaking the input detection in the game by testing the SDF mode with different number of sliders declared in the HID Descriptor (all sliders values were zeroed so there was no interference with slider activity):

- No Sliders: buttons register correctly
- 2 sliders: buttons register correctly
- 6 sliders: buttons register correctly
- 7 sliders: triangle button does not register anymore
- 8 sliders: triangle and L2 don't register anymore.

It seems after the 6th new slider each new one included breaks a different button, though more tests weren't made to fully determine if this is the case.

# Conclusion

It seems the number of sliders is the one thing breaking input detection in-game. Though this behavior was only verified with a single game, it may also happen with other old games. The best route to keep compatibility appears to be a new HID Device mode based on SDF but without the pressure sliders (`Common Gamepad` Mode).

MetalGearAlex1, 2023-02-15:
Hello. I am having this exact issue. I am reading thru all ur stuff here and there in these forums. What exactly do i need to do to fix my ps3 cobtroller to work properly on pcsx2 ver 1.7049 nightly build. For MGS3 Subsistence. I just need to know if there is a fix. Please!

nefarius, 2023-02-16:
On latest PCSX2 Qt builds simply use SXS instead, they recently added support for that.

MetalGearAlex1, 2023-02-16:
> On latest PCSX2 Qt builds simply use SXS instead, they recently added support for that.

Ok. Ill try that. Thanku!

nefarius, 2023-02-16:
IDK, I didn't make the change nor have had the time recently to fire up PCSX2, you best search their issues trackers for the related issue.

MetalGearAlex1, 2023-02-16:
> IDK, I didn't make the change nor have had the time recently to fire up PCSX2, you best search their issues trackers for the related issue.

Ok. So i will do that. But i got a ps3 controller for the pressure sensitive buttons & MGS3/2

closed 2026-09-20 by nefarius