nefarius/DsHidMini #77: all buttons will be pressed when switch windows

Mr-Beacon00, 2021-04-24:
all sliders value is 128 when switch windows.
when ds3 connected is will happened too
sorry for my English. problem just like this video.
https://www.youtube.com/watch?v=EEyrLsEoq-4
[image]

nefarius, 2021-04-24:
Hello,

This seems like an issue with PCSX2 since the driver doesn't care which window is active. Can you try with different controllers or a newer version of PCSX2?

Cheers

Mr-Beacon00, 2021-04-30:
I have tried 1.4.0 1.6.0 and 1.7.0 version. is all same
using "Monitor when in background" is no problem.
rpcs3 don't have problem like this

nefarius, 2021-04-30:
Looks like a PCSX2 bug to me. The driver always reports the current state of the controller, so this behaviour is exclusive to PCSX2. Please report the issue to them, nothing I can do about here.

Cheers

closed 2021-04-30 by nefarius