nefarius/DsHidMini #78: This program broke every single control scheme in every single game I have

Zenostop, 2021-04-25:
Some games select is O and back is square, other games select is triangle and back is X? Both Steam and non-steam games are messed up. Changing the HID modes just changes it to other incorrect buttons. I'm going back to SCPToolkit. It worked perfectly.

Kanuan, 2021-04-25:
Good riddance. That's what happens when you use a software that you don't know how it works.

If you had took 5 min of your time to try looking for answers, you would know that [DsHidMini works differently than ScpToolKit](https://github.com/ViGEm/DsHidMini/issues/40), requiring extra steps to achive xbox360/ds4 emulation

Zenostop, 2021-04-25:
99% of people here don't know how this works. I just wanna plug in my controller and play. Thanks for the link.

closed 2021-04-25 by Zenostop

Kanuan, 2021-04-25:
There are 4 different links on the download page of DsHidMini instructing how to use DsHidMini.

If the user don't want to read then it's not our fault.

If the user wants to use ScpToolKit, a bad and obsolete program, just because it's "easier", it's also not our fault.

Zenostop, 2021-04-26:
How do I hide HID DS3 Compliant Device? Using HidHide hides it from DS4Windows and doesn't allow me to use it.

Kanuan, 2021-04-26:
There's a section on it on the [DS4 Mode User Guide](https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/#hiding-the-controller) or you can use the [generic HidHide guide](https://vigem.org/projects/HidHide/Simple-Setup-Guide/#allowing-chosen-applications-to-see-hidden-devices).

The next version of DsHidMini won't require HidHide anymore, the DS4 Mode will be invisible by default on itself

Zenostop, 2021-04-26:
Restarting my PC fixed it. Thanks.

nefarius, 2021-04-26:
> 99% of people here don't know how this works. I just wanna plug in my controller and play. Thanks for the link.

Really? 99% of the over a thousand downloaded copies ended up with users who can't read? Interesting how the math works out.

It is plug and play, just not the way **you** expect it so stop whining and assuming your case is the only true way.

I didn't create this as a 1:1 SCP successor, I created it primarily for compatibility with **emulators**, the XInput stuff is just a convenience extra.