nefarius/DsHidMini #81: Request: Disable digital face buttons on Steam profile

netracerx, 2021-05-05:
Certain games, like TDU2, do not see the analog face buttons when the Steam profile is used. Attempting to use any other profile does not seem to configure the controller consistently.

nefarius, 2021-05-05:
Can you elaborate what you mean by "Steam profile"? And in general more information how you have your setup configured, I don't quite understand what is going on here, thank you!

netracerx, 2021-05-05:
The SXS (Steam/RPCS3) profile does not have the ability to mute digital 
pressure buttons to give the analog pressure functionality priority. I 
use the face buttons (pressure buttons) for things like accelerate, 
brake, and e-brake in racing programs as that's much quicker for me to 
use than the trigger buttons.

nefarius, 2021-05-05:
I see. Yes, that is on purpose; this mode exactly emulates the behaviour that the Sony `sixaxis.sys` driver uses and Steam is "tuned" to. In this mode it is not possible to access the pressure information, unless you use a custom request like RPCS3 does.

I just tested it; I can not remap the pressure axes in the Steam Controller UI and I also see no packets for fetching pressure data in the driver tracing logs.

This is already explained [in the documentation](https://vigem.org/projects/DsHidMini/HID-Device-Modes-Explained/#sxs) and works as intended. If you want to utilise the pressure axes, use SDF or GPJ.

netracerx, 2021-05-05:
I've tried to use SDF, and I not only do not get the pressure sensitive 
buttons, but the cross/POV buttons now take over analog inputs. Also, 
GPJ kinda works, but because it's breaking the controller into two 
directinput controllers, it confuses games like Test Drive Unlimited 2 
or The Crew, which expect either a single controller for input or input 
from the keyboard/mouse.

nefarius, 2021-05-05:
This is - again - expected; the layout of the other modes are artificial, I "invented" them. They are meant mainly for the **suggested** programs in the UI and the docs. SDF definitely supports and exposes the pressure information **if** the process in question uses the HID-API and not DirectInput. For DirectInput compatibility due to axis limits you need to use GPJ.

None of the Games or Steam will probably recognize the gamepad layout, because, again, it is artificial. You need to use the remapping capabilities of Steam or the game or a 3rd party remapper for that.

To mitigate the "2 devices confusing the game" you can block the 1st device using [HidHide](https://github.com/ViGEm/HidHide).

netracerx, 2021-05-05:
Thanks anyways, I was trying to see if I could get the same 
functionality that was available from the MIJ (MotionInJoy) 
application/driver, but it does not seem like this is forthcoming from 
this project. Again, thanks.

Kanuan, 2021-05-05:
Strange that Steam recognizes GPJ pressure buttons but not SDF. I will try investigating this when I have time.
For now, you could try using SDF + UCR to create a virtual Xbox Controller where the triggers are actually some pressure buttons

nefarius, 2021-05-05:
> I was trying to see if I could get the same functionality that was available from the MIJ (MotionInJoy)

Can you explain in detail what MIJ did? If my memory serves me right it presented the DS3 as an Xbox Controller, correct? If so, how were the pressure axes actually "used"?

Thanks

Kanuan, 2021-05-05:
If I recall correctly, it allowed the remap a pressure button to one axis (triggers or sticks) of the emulated Xbox 360 controller

netracerx, 2021-05-05:
Actually, that was ONE mode it worked in. It's primary mode was a 
directinput mapper for the DS2/DS3, allowing a user to map any virtual 
button or axis to any physical button or axis. It was pretty neat in 
that it did allow you to unmap axes you didn't use (the gyro, for 
instance) and remap those elsewhere, as well as splitting axes so you 
had more available to use.

nefarius, 2021-05-05:
> Actually, that was ONE mode it worked in. It's primary mode was a directinput mapper for the DS2/DS3, allowing a user to map any virtual button or axis to any physical button or axis.

Ah, so it sort of created a "customised HID device" similar to what happens here. Well, if we knew how exactly it was done I am very certain we can just replicate it.

I am a bit concerned to directly add remapping logic to the driver because that overhead is not to be underestimated but it can be done.

Maybe someone brave installs MIJ again and we dump the used HID Report Descriptor 😆

netracerx, 2021-05-05:
Maybe if I knew how, I'd do it just to help out. I will say, BetterDS3 
is the offline configurator for the MIJ driver, designed for those that 
did NOT want to use the online configuration utility. Just FYI.

nefarius, 2021-05-05:
A "trustworthy" copy and a Virtual Machine will do the trick, we shall use this issue to keep track 👍

netracerx, 2021-05-05:
I'll have to see about that, then. I have an old copy still from 2016, 
but the driver won't install unless you set the clock back to some 
stupid time before the cert on the driver expires. I'd have to look all 
of that up.

nefarius, 2021-05-05:
That is not an issue. If you have a link, share please. I meant that **I** can spin up a VM, not you. After all I have all the sniffing tools already around 😉 And I can test-sign it so the kernel loads it, no biggie. I think IIRC it worked best on Windows 7.

netracerx, 2021-05-05:
[MotioninJoy_071001_signed.zip](https://github.com/ViGEm/DsHidMini/files/6430782/MotioninJoy_071001_signed.zip)
Attached, hopefully, is the last version that was put out. Should be safe to load.

netracerx, 2021-05-06:
Yeah, the original author left the project, and someone took it over, then did that to it. Sad times. Was such good software.

nefarius, 2021-05-06:
> Yeah, the original author left the project, and someone took it over, then did that to it. Sad times. Was such good software.

Nice way of quickly and efficiently nuking your reputation. But I digress, it works in the VM so I can experiment with it 😉

PR #219 (2022-04-26): Introduces new JSON-based configuration system
## What's changed

- Fixes #196 
- Fixes #81 
- Fixes #53 
- Fixes #52 
- Fixes #51 
- Fixes #35 
- Fixes #30 

## What's missing

- DSHMC needs a total overhaul as this change is 99% incompatible with the current config system
- See open tasks in #197 for open work
closed 2022-04-26 by nefarius