nefarius/DsHidMini #83: ds3 support

SamsungHandy, 2021-05-06:
latest update added ds3 support but can i now use my ps3 controller at the same time as i use my ps4 controller
dont know how to find it

nefarius, 2021-05-06:
... what?

SamsungHandy, 2021-05-06:
Support for DS3 gamepad as vendor defined device over DsHidMini driver. Contribution by mika-n
how do i add my ps3 controller

Kanuan, 2021-05-06:
Your question makes no sense. Github's issue tracker is not a help desk, it is for actual __issues__, not go around asking for help.

Read the documentation, we won't offer help for things already covered on the guides
https://vigem.org/projects/DsHidMini/
https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/

nefarius, 2021-05-06:
You're speaking about DS4Windows then? Can you maybe use the basics of human communication and whip out more context than meaningless unstructured sentences?

DS4Windows already has support for DsHidMIni, the recent change means it will now support a - not yet release - version that will no longer required HidHide.

I'll close and lock this nonsense.

closed 2021-05-06 by nefarius