nefarius/DsHidMini #86: On `DS4Windows` Mode, the `pairing attempt` will always be marked as failed

Kanuan, 2021-05-11:
On the new, yet unreleased DS4Windows mode, the last pairing attempt will always result in a fail, at least according to DSHMC.
Don't know if the pairing is actually failing or if DSHMC is just failing to properly check its status

[image]

nefarius, 2021-05-11:
That was probably me during the last major refactoring run, I remember touching that function. I'll look through the differences and see what I can do. Thanks!

nefarius, 2021-05-11:
Oh, or is it really mode-dependant?

nefarius, 2021-05-11:
Can you check the property what the exact error code is in your case?

[image]

Because I got it now as well while plugging it in (in new XInput mode):

[image]

Kanuan, 2021-05-11:
> Can you check the property what the exact error code is in your case?

It's the exact same as yours

nefarius, 2021-05-11:
> > Can you check the property what the exact error code is in your case?
> 
> It's the exact same as yours

Then I am quite certain this is the case. I added the check in the refactoring, to avoid re-pairing to the same radio which avoids sending an unnecessary packet.

closed 2021-05-12 by nefarius