nefarius/DsHidMini #87: Request:  Option in Settings to have LEDs show P# rather than battery/charging status

johnodon, 2021-05-13:
My controllers sit on charging cradles all day so I (personally) don't have much of a need to know current battery status. I have young boys who love to put their controllers down to go to get a drink or go to the bathroom (or something) and come back and say "which controller is mine?".  If the LED displayed the P# I won't have to truck upstairs to solve their mini mystery. 

Certainly a low priority request but, thoughts?

John

nefarius, 2021-05-13:
A somewhat realistic player index indicator is only possible in pure XInput mode. All other APIs have no concept of tying device and player slot together. It mostly depends on the connection order, the APIs the game use and more which is opaque to the common user.

I can keep track of controller indices internally to calculate a potentially realistic player index based on connection order but that breaks as soon as different models (Xbox controllers, other 3rd party devices) are used together on the same system.

All methods I can think of are prone to error and will lead to confusion and support inquiries without and understanding of the core issues.

I can offer a setting per controller where the user itself can request a fixed LED lit all the time for a particular controller. I am against an automatism for the reasons outlined above.

Thoughts welcome ofc.

Kanuan, 2021-05-13:
Instead of DsHidMini specific, a more general purpose tool would be more interesting?

I was thinking something along the lines of UserScp.exe, a small tool that can show the current xInput indexes, but stayed in system tray and could be called as a button combo (Xbox + RS, for example) and created a temporarily overlay showing controller inputs per slot

Rast1234, 2021-08-03:
Just an assignment like `device ID/MAC/??? => LED pattern` would be nice, persistent and editable. I suppose most people don't have 10s of controllers, so it would be enough to tell one controller from another at any time. It can be automatic in means that when DsHidMini learns new ID, it assigns incremented LED pattern, i see no downside to this.

BTW i do have a lot of controllers for different systems used together and ended up attaching small stickers with phonetical names like Alpha, Bravo, Charlie, Delta. But LEDs are more fun of course!

nefarius, 2021-08-03:
A truly unique method to keep these settings persistent is only possible on wireless anyway, via USB these controllers offer no means of uniqueness so that Windows would always consistently assign them the same "slot".

Rast1234, 2021-08-03:
IMO it doesn't matter if they stay in same slot when changing connection method, it's just about finding your controller after you left computer for 5 minutes :)

nefarius, 2021-08-03:
> IMO it doesn't matter if they stay in same slot when changing connection method, it's just about finding your controller after you left computer for 5 minutes :)

That is **your** opinion, I'm sure as soon as it's implemented, other support requests will pour in like "why doesn't it behave exactly like my niche case?!" 😉

Kanuan, 2021-08-04:
@Rast1234 if you don't depend on SDF/GPJ/SXS modes specific features and the games/emulators your kids are playing support Xbox controllers you could more or less achieve what you want with DS4Windows.

It can't tie the LEDs to the XInput slots, but on DS4Windows you can control the controller's LEDs by setting specific lightbar colors on the aplication.

You would need to create 4 different profiles, each one with a different LED (lightbar color) set in its settings and then you could tie the profile to the controllers' ID/Mac

The correct lightbar colors for LED control are found on the [DS4Windows User Mode Guide](https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/)

 
[image]

vertigo220, 2021-08-10:
Why not jut go low-tech and write their names on the controllers so they always just use their own?

Rast1234, 2021-08-11:
> Why not jut go low-tech and write their names on the controllers so they always just use their own?

Of course i did exactly that :)

> ended up attaching small stickers with phonetical names like Alpha, Bravo, Charlie, Delta

nefarius, 2022-02-15:
> Just an assignment like `device ID/MAC/??? => LED pattern` would be nice, persistent and editable.

This is coming with the next release.

closed 2022-11-06 by nefarius