nefarius/DsHidMini #88: Vibration not working over bluetooth, but working through usb

rissei, 2021-05-18:
BthPS3 1.3.127.0
DS4windows 3.0.5
Dshidmini v1.4.222.0
Intel wireless bluetooth USB\VID_8087&PID_0A2B&REV_0001

After following the main guide and setting everything up, I noticed, using MightyHIDTest, that the rumble funtion does not work over bluetoth for my ps3 controller, is this a resurgence of #4? also apparently according to Dshidmini my controller is not genuine.
[image]

rissei, 2021-05-23:
Umm, hey?, I still have this problem, and need assistance

Kanuan, 2021-05-23:
Oh, forgot about this. 
It's not related to the issue you linked, that was (supposed to be) fixed for good.
Can you send a screenshot of your DSHMC when connected over BT?
Besides that, join the server's discord so it's easier to test on this

rissei, 2021-05-23:
Alright, will upload once I get to my laptop, do I send it here or to the discord server?

Kanuan, 2021-05-23:
send it here to keep it registered, then we continue on the server

rissei, 2021-05-23:
[image]
[image]

Done!

closed 2021-05-31 by nefarius