nefarius/DsHidMini #89: Buttons not responds properly

Marco-Buonaguidi, 2021-05-18:
BthPS3 1.3.127.0
Dshidmini v1.4.222.0
PCSX" ver. 1.6.0

I have followed the installation guide and i've tried to use the PS3 controller on PCSX2. The mapping of the buttons on PCSX2 work fine and all buttons are correctly recognized and mapped, but when i start a game the buttons doesn't respond properly. For example, the start button doesn't work but if i press L2 button it work like start button and L2 button at same time, as if both inputs are sent to the emulator. Accordingly to dshidmini, my controller is genuine.

Kanuan, 2021-05-18:
1. Send a screenshot of the DsHidMinic Control Cliente (DSHMC)
2. Is your controller in SDF mode?
3. Check the [_How do I set the controller's pressure buttons in PCSX2?_](https://vigem.org/projects/DsHidMini/FAQ/#how-do-i-set-the-controllers-pressure-buttons-in-pcsx2) topic in the FAQ

Marco-Buonaguidi, 2021-05-19:
Hi,

i've attached the requested screenshots. I've followed the guide and i've set the SDF mode and used the "Mute digital pressure buttons" to configure the pressure sensitive buttons. The problem persist, even if i connect the pad via usb, there isn't any difference.
If can help, the controller that i'm using is a SIXAXIS, not a DUALSHOCK 3. Is the first version of the controller, that was without vibration.
I also have another controller DUALSHOCK 3 that is reported as fake and it has the same problems.

Let me know if you have necessity of other information to debug the problem, meanwhile i'll do other tests with other versions of PCSX2 too.

nefarius, 2021-05-20:
I'm afraid GitHub strips away any attachments if you reply via mail rather than the Web browser so I'm afraid we can't see what you see 😊

Marco-Buonaguidi, 2021-05-20:
Sorry

[image]
[image]
[image]

Marco-Buonaguidi, 2021-05-22:
Hi guys,

after other tests i can say that the problem occur only with some games. I've tried to play with Gran Turismo 3 and Gran Turismo 4 and both have the problem, but i can play Metal Gear Solid 2 normally, the problem doesn't occur on this game.

nefarius, 2021-05-22:
> Hi guys,
> 
> after other tests i can say that the problem occur only with some games. I've tried to play with Gran Turismo 3 and Gran Turismo 4 and both have the problem, but i can play Metal Gear Solid 2 normally, the problem doesn't occur on this game.

Just to clarify; we're still speaking exclusively about games running PCSX2, correct?

Marco-Buonaguidi, 2021-05-22:
Yes, i've used the pad only with PCSX2 at the moment.

nefarius, 2021-05-22:
How odd, once the mappings are set up, I see no logic in why games would respond differently to the emulator configuration...

Marco-Buonaguidi, 2021-05-26:
I don't know if could help, but i have a similar problem with a Logitach G923 racing wheel. During the configuration of the input via the plugin on PCSX2 all seems to work properly, but in the game (gran turismo 3) the inputs are wrong, like the problem with the pad. Could be a problem of PCSX2?

nefarius, 2021-05-26:
Guess I need to take a look at the LilyPad sources to understand what's going on.

closed 2021-05-31 by nefarius