nefarius/DsHidMini #90: Windows (sometimes) recognizing the controller in DS4Windows mode as an gamepad

Kanuan, 2021-05-19:
## Summary

Since in DS4Windows Mode the controller is supposed to be remapped into a virtual controller by DS4Windows, to prevent the famous double-input issue the controller's [HID descriptor was changed](https://github.com/ViGEm/DsHidMini/pull/71) so Windows stopped recognizing it as a gamepad.

Some users are reporting that in the latest stable driver Windows still recognizes the controller in DS4Windows Mode as an actual gamepad.

## Details

From the image below, Windows is ignoring the X, Y, Z, RZ axes and the T-Hat, though RX and RY as well as the other buttons are recognized as normal  :
[image]

Looking at the HID descriptor, it seems as windows is ignoring everything _before_ the usage page is refreshed:  
[image]

## Test

I wanted to confirm if the issue was being caused by windows recognizing the axes/buttons after the usage page was "refreshed" in the HID descriptor, so I replaced all gamepad related sections of the HID descriptor (all axes, buttons, t-hat) with vendor-defined padding:
[image]

And the result was that Windows __still__ recognizes the controller as a gamepad, but now without any common feature:  
[image]

## Final thoughts on the issue

I can only think that Windows got stuck identifying the controller as a gamepad, regardless to what the initial usage and usage page is defined on the HID Descriptor. I don't know how to further troubleshoot this, since I can't replicate this on my system.

nefarius, 2021-05-19:
Since DS4Windows has its own byte-based parser anyway I think the whole Report Descriptor could be further simplified by not exposing any "well-known" usages all together, just make sure the report sizes still match, nothing else so the hidclass parser has nothing to expose to the system.

nefarius, 2021-07-02:
@Kanuan could you test my change please? I basically ripped out all known usages and just fixed the overall padding of the input report, at least on my end DS4Windows (`3.0.11`) seems to be happy with the change and still functions while the original devices remains fully hidden:

[image]

commit ea6b2eb3 (2021-07-02): Resolves #90
closed 2021-07-23 by nefarius