nefarius/DsHidMini #113: Some games recognize controller in XInput HID Device Mode as 2 different controllers

Kanuan, 2021-07-25:
## Summary
In the new `XInput HID Device Mode` introduced in DshidMini v2.0.242, the controller acts as a DInput device with a XInput compatibility layer added on top of it.
Games that don't support generic controllers will recognize the DS3 as just an Xbox One controller, and most games that do support both generic and XInput controllers can detect that both the DInput and Xbox controller are supposed to be one, therefore ignoring the DInput part of it.
The problem is that some games will still detect the DS3 controller in `XInput HID Device Mode` as 2 separate devices. Depending on how the game behaves with generic controllers this behaviour can make the game unplayable.

## Reproducing the issue
- Set the controller in `XInput HID Device Mode`
- Use the controller in one of the games that can't detect that both the DInput and Xbox "parts" of the `XInput HID Device Mode` are supposed to be one controller

## Known affected games
- Need For Speed Most Wanted (2005) - Unplayable
- Tekken 7 - Unplayable? (Requires more testing)
- Enter the Gungeon (Game will detect the DInput controller but will not use it by default)

## Ways around the issue / Notes on what has been tested
- Probably the only way around this is by .dll/api hooking that removes DInput support for the affected games (untested)
- It's impossible to hide the DInput part of the controller with HidHide since the controller doesn't appear at all on HidHide's device list
- HidGuardian was not tested
- Steam's `Xbox Configuration Support`/`Generic Configuration Support` does not prevent the issue from happening when a game is launched through Steam

nefarius, 2021-07-25:
I don't think this deserves an issue. This approach is experimental from the very beginning, so whatever is not working is to be discovered via discussions and if concrete details bubble up on what can exactly be improved upon the driver, that would then warrant an issue, but not really a general "some games something something", I was aware we're not gonna have a clean solution due to the sheer amount of shit game engines out there. My 2 cents for now 😜

nefarius, 2021-07-25:
Another thought that came up: it's nice and helpful that we start collecting information about which games work and which are troubled, but **as a developer what am I supposed to do with this information**? I'll never have all the mentioned games myself to test neither the time to invest. So, what would be the end goal? A database of "if this game better use DS4Windows route"? The new approach may not be as universally perfect as simply emulating the good old x360 pad (which is **impossible** natively with DSHMs architecture) but it opens new ways to explore how XInput is processed in games and what to do to improve it.

So TL;DR: what to do with this information and what can be given to the end-user to easy diagnostics?

Kanuan, 2021-07-25:
> I don't think this deserves an issue. This approach is experimental from the very beginning, so whatever is not working is to be discovered via discussions and if concrete details bubble up on what can exactly be improved upon the driver

I moved all the relevant info to the [discussion topic](https://github.com/ViGEm/DsHidMini/discussions/114)

I answered your other thoughts there too.

closed 2021-07-25 by Kanuan

BlueStarBRS, 2026-02-18:
I swear, it's pure horror that HidHide doesn't show the DsHidMini input devices. That's literally the biggest reason not to use DsHidMini to finally replace the SCP Toolkit. There are other scenarios, too, where this just sucks  like in games that detect both, or when you really need to hide your input sent from DsHidMini (like in my case with a modular controller emulation translation layer)
(discord joined... waiting for verification to find a modern solution▸#newbie-8487)