nefarius/DsHidMini #118: Lilypad on PCSX2 won't allow different SDF DS3 controllers to be mapped to different PS2 pads/ports simultaneosly

TSnatch72, 2021-07-27:
On SDF mode for PCSX2, Lilypad (the controller plugin) won't allow one DS3 to be mapped to pad 1 port while mapping the other one to pad 2 port. The result of attempting to do so is Lilypad deleting pad 1 bindings when binding the same button to any input on pad 2, treating all DS3 controllers as single one and therefore only allowing all DS3 controllers to be mapped to a single PS2 controller. 

The only way i have seen to be able to map DS3 controllers to Pad 1 and Pad 2 simultaneously is to enable the "multiple bindings" option, but that would only make any DS3 to control both pads at the same time ala hive mind, so it wouldn't be usable for 2-player games

nefarius, 2021-07-27:
Interesting, I'll test on my end too, haven't tried with more than one controller I must admit! #sadSoloGamer

TSnatch72, 2021-07-30:
I feel you hahahahahahah
don't really play with other people locally often either.

nefarius, 2022-06-28:
A reliable workaround is using SXS with the SCP Proxy DLL.

closed 2022-06-28 by nefarius

TSnatch72, 2022-06-28:
Does it support pressure sensitive buttons through the dll?

nefarius, 2022-06-28:
Yes and yes.

TSnatch72, 2022-06-28:
So if i'm getting this right, the scp proxy dll is the xinput dll included in DsHidMini, right? So the workaround is placing the dll in the pcsx2 folder and then setting the controller as SXS