nefarius/DsHidMini #119: Cannot install to Windows 11 beta

fabiograssiotto, 2021-07-30:
Hi,

Windows 11 beta version 10.0.22000.100 is available for Windows Insiders testing.
When migrating from Windows 10 the driver is failing to be initiated.
Right clicking on the inf file and selecting 'install' does not bring the popup window as described on https://vigem.org/projects/DsHidMini/How-to-Install/

Thanks,

nefarius, 2021-07-31:
Oh dear, Windows 11 incoming 😅 Please look for `C:\Windows\INF\setupapi.dev.log` and attach it here, thanks!

fabiograssiotto, 2021-07-31:
[setupapi.dev.log](https://github.com/ViGEm/DsHidMini/files/6911750/setupapi.dev.log)
Attached. There is probably some more stuff going on on the file, I installed SCPToolkit for now.

Thanks!

nefarius, 2021-07-31:
According to the log the driver installed successfully. Make sure to really wipe every trace of SCP or other conflicting solutions, as per documentation. I can not see anything obvious why it shouldn't work.

closed 2021-08-04 by nefarius