nefarius/DsHidMini #132: Triggers not working in Sonic & All-Stars Racing Transformed, in XInput mode.

dxgldotorg, 2021-08-28:
- [x] I have searched open and closed issues for duplicates
- [x] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
Trigger buttons not working in Sonic & All-Stars Racing Transformed (Steam App 212480) when in XInput mode.  Looking at the Steam forums this might be an issue with XBOX One controllers too, so I tried the suggestion of disabling XBOX controller support in Steam, no luck.

### Steps to reproduce
- Install DsHidMini 2.0.248 and its drivers.
- Set controller to XInput mode
- Start a game of Sonic & All-Stars Racing Transformed

**Actual result:** Triggers have no effect, and player's car doesn't move.
**Expected result:** Right trigger should accelerate, and left should brake.

### Machine info

**CPU-Architecture:** x64
**Windows version:** Windows 11 build 22000.168 Dev
**Software/driver version(s):** 2.0.248

### Any other helpful information
Tested in both BT and wired.

dxgldotorg, 2021-08-28:
Closing due to known issue at https://github.com/ViGEm/DsHidMini/discussions/114

closed 2021-08-28 by dxgldotorg

dxgldotorg, 2021-08-28:
Fixed by force enabling Steam Input per-game.

nefarius, 2021-08-28:
> Fixed by force enabling Steam Input per-game.

Yesssss worked for me too in the game Road Redemption!