nefarius/DsHidMini #133: DS3+DsHidMini+DS4Windows Combo Not Working

BigheadSMZ, 2021-08-29:
- [x] I have searched open and closed issues for duplicates
- [x] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
Using a PS3 controller, DsHidMini with HID device mode set to DS4Windows, and using DS4Windows itself with Emulated Controller set to "XInput", the controller is detected as "Controller (USB Controller)" and does not work with anything that uses XInput. Changing the emulated controller to "DualShock 4" makes it appear as "Wireless Controller", and again, does not work with anything.

### Steps to reproduce
I followed the guide here, step by step:
https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/

**Expected result:**
I expect the controller to properly show up as an XInput controller. Instead it's showing up as some generic name "Controller (USB Controller)" and doesn't work with anything that is compatible with XInput. The readouts work in DS4Windows, the readouts work in Windows "Game Controllers" window, but the controller doesn't work as XInput. I used DS4Windows "XInputChecker" and it says connected, the buttons are working here. So I am at a loss, I have no idea why it's not working.

### Screenshots
[image]
[image]
[image]

### Machine info
**CPU-Architecture:** x64
**Windows version:** Windows 10 20H2
**Software/driver version(s):** 1.4.222.0

BigheadSMZ, 2021-08-29:
I guess I'm an idiot, the reason it wasn't working is because it was appearing as controller 2. I have no idea why as I only had my PS3 controller plugged in when setting it up and testing. I'm guessing it had to do with some settings in DS4Windows, there was an option to disable it showing up as 2 controllers instead of 1. After multiple restarts and much fiddling, it's now working correctly despite just showing up as "USB Controller". I gotta admit, this is a bit cumbersome to use in it's current state due needing either DS4Windows or X360CE, and I was tempted to revert back to SCP Toolkit multiple times throughout this endeavor since it's just install, plug, and play, while this required jumping through a bunch of hoops (multiple driver installs, multiple programs running, multiple settings). 

Either way, great work on this and I'll continue using it for now at least. Another issue I was experiencing adding to my frustration is that DsHidMini was picking up my Retro Fighters Brawler64 controller as a "PS3 Controller" and took it over completely. This disabled it's own ability to switch between XInput and DirectInput, and it did not work using DS4Windows. It took me a bit to figure out that I needed to manually go into Device Manager and uninstall the "Nefarius" driver for this specific controller to restore it to normal functionality. I also noticed that XInput is planned to be built into DsHidMini eventually as an "HID device mode", if that eliminates the need for DS4Windows/X360CE, that would greatly enhance this program's usability.

closed 2021-08-29 by BigheadSMZ

nefarius, 2021-08-29:
> I guess I'm an idiot, the reason it wasn't working is because it was appearing as controller 2. I have no idea why as I only had my PS3 controller plugged in when setting it up and testing. I'm [...]

IDK what happened there but if there is no interfering legacy drivers, installation is dead easy and done in like 30 seconds. And requiring multiple programs is the design decision for a modular architecture, which is the sole reason this solution is able to exist, another monolith like SCP would've probably required another 5 years to design and probably be abandoned due to frustration. **TL;DR:** it is how it is and it is better that way 🙂 for **me** as a developer which is the primary concern, the user has to spend 30 instead of 10 seconds to set it up? So sad 😆  /SCNR

> Either way, great work on this and I'll continue using it for now at least. Another issue I was experiencing adding to my frustration is that DsHidMini was picking up my Retro Fighters Brawler64 controller as a "PS3 Controller" and took it over completely.

Complain to the manufacturer of that thing way they spoof Sony's Hardware IDs then?! Not my problem 🙂 

> I also noticed that XInput is planned to be built into DsHidMini eventually as an "HID device mode", if that eliminates the need for DS4Windows/X360CE, that would greatly enhance this program's usability.

Mate, it's literally out there for a month now in v2 Beta, check the releases 😅 

Cheers, and game on! ❤️

BigheadSMZ, 2021-08-29:
If you are just creating this for yourself and sharing it with the public as a bonus, I understand that. Most creators do. But such a hostile reply when I was only trying to help by reporting what I thought was a bug, and when figuring it out, offering constructive criticism from a user's standpoint. And in the end I admitted that the fault was mine. Maybe it "takes 30 seconds" to set up with someone who is familiar with all of this, like you who spent years researching and creating it, but never using any of this before requires at least some level of research and understanding. And that is not a problem, I'm fine with learning new things, I like when programs offer more power. If someone who has never used this or DS4Windows, can read over and understand every single option in both programs in 30 seconds, especially when they run into issues like I have, then yeah that is impressive. Kudos. Duh, I dumb, you smert (insert sarcastic face emoji of your choice).

nefarius, 2021-08-29:
My replies were in no way intended as hostile or insulting your intellect, but since you already decided to interpret in that way I guess there's no reason for further discussion... Nonetheless, have a fabulous day!

BigheadSMZ, 2021-08-29:
When you quoted my reply, talked about how easy it was, and ended with "So sad/laughing face/SCNR", it appeared as though you took a shot at me. I do apologize if I misinterpreted it, I honestly didn't come here to start trouble. I really am grateful for your work and efforts, I would have given up if immediately and never made a report at all if I wasn't interested in it.

nefarius, 2021-08-29:
No need to apologise, mate, text communication is hard for all parties involved. My sarcastic or cynical comments are typically thrown out as a generalisation at past experiences, it is not my goal to make any individual look bad or stupid, especially when they get use out of my chaotic creations 😜