nefarius/DsHidMini #138: certificate error on Windows 11

synapp009, 2021-09-14:
When I tried to install the dshidmini.inf I got following error on Windows 11:

dshidmini.inf: Failed to add driver to the system. Error 0x800B0110: The certificate is not valid for the requested usage.

nefarius, 2021-09-15:
Have you recently applied a Windows update? We had another user on the Discord where a recent update ruined the included root certificate chains. This is a Microsoft issue, nothing I can do about. The certificate on the driver is perfectly fine and works throughout all Windows 10 builds as expected.

synapp009, 2021-09-15:
It happens with the new Windows 11 build 22454.
So it might only be somemthing to worry in the future.
Thanks for your fast response.

nefarius, 2021-09-15:
I don't really consider Windows 11 "stable" and "mainstream" yet, Microsoft is constantly changing the rules and break things in a subtle way, independent developers can't keep up with this nonsense. I will keep it on my radar but for now I consider DsHidMini **stable** and Windows 11 **beta**, not the other way around 😛 

Cheers

nefarius, 2021-10-01:
Alright, it has been confirmed this is an issue with a certain Windows 11 update. Unfortunately we haven't taken note of which specific one, if someone knows, please chime in as it seems a folluw-up update has fixed the issue.

Cheers

nefarius, 2021-10-01:
Got notice from a user that it is back in the currently most recent build 🤣 

[image]

Maybe Microsoft shouldn't have fired the entire QA department... Always nice to see OS **basics** break randomly.

Jjagg, 2021-10-01:
Ah, I just worked around this before updating Windows by temporarily [disabling driver signature checking](https://www.howtogeek.com/167723/how-to-disable-driver-signature-verification-on-64-bit-windows-8.1-so-that-you-can-install-unsigned-drivers/).

nefarius, 2021-10-01:
I'd love if for a change Microsoft would swiftly fix **core components they casually break all the time** though 🤣

chrisdlees, 2021-11-14:
I think @Jjagg is right, the updates are actually going out of their way to break the bluetooth driver. Might be Win11's BTH radio management, because mine has been behaving quite bizarrely. The past month, I've seen 5 updates cause my bluetooth connectivity to magically disable/re-enable.
   Right now, my wireless is working, but I've had the latest update standing by for over a week. I'm avoiding it at all costs, even going so far as to switch to working in a WSL2 dev environment or RDPing into an AWS EC2 instance, instead of rebooting into my normal linux installation... lol

nefarius, 2021-11-15:
Isn't that exactly what one wants for a productive machine? Stuff randomly breaking? 🤣 I'll stay on 10 for as long as possible, public Beta products aren't my thing 😉

nefarius, 2021-12-08:
There's news on this topic, apparently Windows 11 will start to mandate Microsoft signed UMDF drivers as well, so future releases will be made "Windows 11-ready" accordingly.

sambolee, 2022-11-22:
Hi there. I've had this error message as well although I am trying something a little different. I am using my macbook pro m1 with parallels and trying to install it. I'm not sure if that makes any difference. Is there a solution for this yet?

Thanks.

nefarius, 2022-11-22:
If there was, this issue would hold said information and would be closed 😉

sambolee, 2022-11-22:
Haha right...I will wait patiently :)

HeavyManCrush, 2023-05-23:
Hey guys, was this ever solved? I'm getting this "The certificate is not valid for the requested usage." issue as well. Using Windows 11 Home build 22621.1702 with latest updates, via Parallels on an M1 Mac Air

nefarius, 2023-05-23:
Nothing has changed in the issue state so no, didn't get resolved.

nefarius, 2023-05-23:
I have a Patreon (linked everywhere, easy to find) but that won't change the fact that I simply have no time for this project for the next couple months 😛

ThomasJeremiah, 2023-11-04:
Getting a variant of this issue,
"Install Error"
"The INF file you've selected does not support this method of installation"
"The required section was not found in the INF."

nefarius, 2023-11-05:
> Getting a variant of this issue, "Install Error" "The INF file you've selected does not support this method of installation" "The required section was not found in the INF."

I mean what are you expecting? The issue is still open and unchanged for a reason...

chrisdlees, 2023-11-06:
I'm a little confused at this point, as mine has been working fine since reinstalling windows 11 from the latest canary/nightly/fastest-ring (I hate how microsoft chooses to name their release cycles) around 6 months ago. I have to uninstall and reinstall bthps3 after each weekly update, but other  than that, no problems...
Perhaps installing from the insider iso, rather than just switching via windows update, enables more than just 'developer mode'....
I sorta misrepresnted this by saying its been "working fine" because after last week, I uninstalled bthps3 BEFORE running the update, and while the uninstallation's --reboot flag was still active. After rebooting, at which time it forcibly updated w11, bthps3 is stuck in a unusable state:
[image]
I'm gonna try using ccleaner or something. Manually removing just the file from the driver store didn't seem to help.

P.S. Im a huge fan of all your work Nefarius, you are a shining star of coding in this cold hard world

chrisdlees, 2023-11-06:
As a tl;dr, the OP and I were/are running insider builds, which were the same major release of w11 as the stable builds of the previous two commenters'. Even prior to cleanly installing the new GPT4/Dali-3-integrated major build, I began being able to use DsHidMini and PS3bth on the w11 installation I first commented about, sometime around it's February insider update. 

Possibly has something to do with being forced to enable developer mode in Windows Settings, or maybe it's a plurality of factors, but switching to insider updates now seems to fix this issue. Otherwise, without a doubt, a:                               

**`Clean install of an up-to-date, October, insider's build of Windows 11 ` _`
WILL`_ ` fix this issue!`** 

Tomorrow, I will try it on a clean install of the current w11 stable build, and attempt to isolate exactly which change to insider builds suddenly fixes this issue.
 
@nefarius Closed: Notabug? Wontfix? Simpleworkaround? Fixesitself?

nefarius, 2023-11-06:
I already said multiple times that a new release needs to be signed through MS Partner Portal, like kernel drivers. There is no time for a new release, hence the situation is unchanged.

nefarius, 2023-11-06:
> P.S. Im a huge fan of all your work Nefarius, you are a shining star of coding in this cold hard world

Thank you kindly, recently though it feels like the fusion reaction is starting to fizzle out. It's not fun living in a world where 99.9% just want want want and the rest is getting shit done. Gets tiresome at some point.

nefarius, 2023-11-06:
> @nefarius Closed: Notabug? Wontfix? Simpleworkaround? Fixesitself?

Consider it not fixed since the way forwards is also submitting user-mode drivers to getting a Microsoft signature.

nefarius, 2023-11-06:
> I'm gonna try using ccleaner or something. Manually removing just the file from the driver store didn't seem to help.

The fix to this is covered in the [BthPS3 FAQ](https://docs.nefarius.at/projects/BthPS3/Frequently-Asked-Questions/#how-to-fix-previous-version-found-on-reinstall) 😛

nefarius, 2024-07-03:
Confirmed fixed in v3 Beta 😎

closed 2024-07-03 by nefarius