nefarius/DsHidMini #143: PS2 to USB adapter is recognized as a genuine PS3 controller in DSHMC.EXE

l3gi0n0fh3ll, 2021-10-01:
- [x ] I have searched open and closed issues for duplicates
- [x ] I am submitting a bug report for existing functionality that does not work as intended
- [x ] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
I bought a cheapo PS2 to USB adapter similar to this a long time ago: [Link](https://www.ebay.ca/itm/324367372894) When i insert it into a USB port on my PC, DSHMC.EXE recognizes it as a genuine PS3 controller.

### Steps to reproduce
- Insert the adapter into the USB port.
-  Open DSHMC.EXE

**Actual result:** Describe here what happens after you run the steps above (i.e. the buggy behaviour)
Fake PS2 to USB adapter is registered as a genuine PS3 controller.

**Expected result:** Describe here what should happen after you run the steps above (i.e. what would be the correct behaviour)
The adapter should be reported as being fake.

### Screenshots
[image]

### Machine info
**CPU-Architecture:** x64
**Windows version:** Windows 10 
**Software/driver version(s):** DsHidMini Control Version: 2.0.248.0

nefarius, 2021-10-01:
If that happens the adapter is spoofing the DS3 Sony Hardware ID so the driver attaches to it, there's nothing I can do against that, thank the manufacturer of that thing for being pricks 🙂

closed 2021-10-01 by nefarius

ryukinix, 2022-11-04:
https://github.com/ViGEm/DsHidMini/issues/143#issue-1013689690
Did you solved in some way that problem? I have the same issue around a shity adaptor for ps2 being recognized as ps3...