nefarius/DsHidMini #182: Add important features to DSHMC

anwar-alsilwy, 2022-02-02:
I hope you can Implement these important functions to DSHMC:

1- Disable/Enable Rumble for all controllers.
2- set default global option for ( HID device mode) for all controllers, even the new ones.
3- add ability to reorder assigned players.

nefarius, 2022-02-02:
- 1. This can't be achieved by the UI alone, needs to introduce an additional setting within the driver, but when done is possible indeed.
- 2. This is not possible in the current design without introduction of some government logic like an always running service that would monitor decide arrival and change settings accordingly. An alternative would be querying a global property on driver startup overriding the default per device setting but that is tricky as an user mode driver has very limited registry access.
- 3. This is not possible.

anwar-alsilwy, 2022-02-02:
Ok,
But please at least let me know how to disable Rumble, it's so important for me.
Thanks

anwar-alsilwy, 2022-02-03:
WOW, okay
I really appreciate your hard work.
Cheers.

closed 2022-02-03 by anwar-alsilwy