nefarius/DsHidMini #183: Bluetooth XInput Bug

hisyam3, 2022-02-04:
While using Bluetooth to connect my DS3 controller (which is genuine btw, this controller came from the ps3 console box itself) and putting it in XInput mode, I found out something weird.

The L2 and R2 actually separated into Z Axis for L2, and Z **Rotation** for R2. And the "PlayStation" button didn't act like the "Guide" button, and just acts like a regular old button.
[image]
In this picture, I actually pressed the "PlayStation" button, and it just activates "Button 11".
<br />

When I plugged it in with the cable, tho...
It works just as intended. The L2 and R2 are both Z Axis, and the PlayStation button actually acts like the "Guide" button.
[image]

I don't know how to replicate it, because I don't know how and/or why it happened.

There's a thing I kinda like about this bug however, and I don't think it would be useful for other old games, but:
Playing Most Wanted with the Widescreen Patch (which adds XInput compatibility) actually makes sense, as L2 and R2 aren't detected by the same Axis, and it makes holding gas while hitting the brakes lightly possible. It wasn't previously possible without the bug, as the patch only detects Buttons and Axis the same way Windows' "Game Controller" settings works.

But at the end of the day, it's still a bug.

nefarius, 2022-02-04:
Your controller isn't in XInput mode on wireless, [see here](https://vigem.org/projects/DsHidMini/XInput-Mode-Explained/#double-check-youre-running-in-the-correct-mode).

Cheers

hisyam3, 2022-02-04:
> Your controller isn't in XInput mode on wireless, [see here](https://vigem.org/projects/DsHidMini/XInput-Mode-Explained/#double-check-youre-running-in-the-correct-mode) and [here](https://vigem.org/projects/DsHidMini/XInput-Mode-Explained/#check-controller-in-joycpl).
> 
> Cheers

Hmm.... I might have forgotten to install the second driver.

closed 2022-02-04 by hisyam3

nefarius, 2022-02-04:
> > Your controller isn't in XInput mode on wireless, [see here](https://vigem.org/projects/DsHidMini/XInput-Mode-Explained/#double-check-youre-running-in-the-correct-mode) and [here](https://vigem.org/projects/DsHidMini/XInput-Mode-Explained/#check-controller-in-joycpl).
> > Cheers
> 
> Hmm.... I might have forgotten to install the second driver.
> 
> Thanks.

No worries, report back once resolved!

Cheers