nefarius/DsHidMini #196: DShdmini Dead-zone threashold feature request

mevin19, 2022-02-13:
Hi, I Would like to request a feature for DShdmini to implement Dead-zone threashold, just like SCPtoolkit global configuration had. With DShdmini my camera/right stick keeps jittering around. I Understand that steam has feature like this that you can just enable but it just doesn't seem to work at all for me, It actually won't work when playing, also when you enable steam input, it adds more latency. Anyways I really appricate this project and would love for this feature to be added, thanks for reading!

nefarius, 2022-02-13:
Sounds reasonable indeed. Before it can become a reality I need to introduce a new way to fetch configuration. Seems like Microsoft hasn't planned a lot of registry access for poor user mode drivers. I'll cook something up in my usual time frame of 4 weeks to 6 months or so 😁

mevin19, 2022-02-15:
> 

Nice! That is awesome, thank you so much :) ! can't wait to make the permanent switch from SCP to DSHD

PR #219 (2022-04-26): Introduces new JSON-based configuration system
## What's changed

- Fixes #196 
- Fixes #81 
- Fixes #53 
- Fixes #52 
- Fixes #51 
- Fixes #35 
- Fixes #30 

## What's missing

- DSHMC needs a total overhaul as this change is 99% incompatible with the current config system
- See open tasks in #197 for open work
closed 2022-04-26 by nefarius