nefarius/DsHidMini #204: Reversed triggers in SXS mode?

Robinde67, 2022-02-24:
Not sure weather the issue is because of the digital trigger buttons or if the trigger axis is reversed, but when using SXS mode and using the trigger axis as input, the trigger register as pressed when fully released (after pressing it once) and remains pressed until pushing down the trigger.

Looking at the configuration settings for the controller, the triggers have two axis, X-rotation and Y-rotation. 
[image]
That's how they look when in idle position. Everything looks visually identical before and after pressing the triggers once.

Not sure if that is the original behavior of the official sixaxis driver, since i can't get them to work at all.

nefarius, 2022-02-24:
That's how `SIXAXIS.SYS` works 🙂

Robinde67, 2022-02-24:
So just Sony being weird and keeping the trigger axis but reversed while having a digital trigger button to mess things up? Yup, sounds legit!

nefarius, 2022-02-24:
I consider this answered then 😉 

Cheers

closed 2022-02-24 by nefarius