nefarius/DsHidMini #211: Create small driver installer

nefarius, 2022-03-21:
I've got enough snippets and libraries now to basically offer a small, simple .NET-based installer, should further reduce the possibility of installation issues either due to missing a step or something unaccounted for, like UAC being turned off.

sitiom, 2022-07-10:
I have added DsHidMini to Scoop for the meantime (https://github.com/ScoopInstaller/Nonportable/pull/22).

nefarius, 2022-08-29:
Progress! Even managed to add a selection-based BthPS3 download-and-install option 🥳 

[image]

gosseux, 2023-07-30:
will this installer appear in the add/remove program list for easy uninstall as well ?

nefarius, 2023-07-30:
> will this installer appear in the add/remove program list for easy uninstall as well ?

Well ofc. 😉

closed 2024-06-24 by nefarius

nefarius, 2024-06-25:
[image]

[image]

PR #367 (2024-06-24): Created WixSharp-based driver installer
Resolves #211