nefarius/DsHidMini #213: XInput Bridge Improvements

nefarius, 2022-04-12:
Notes on user feedback:

- Fix packet counter, it currently wrongly assumes that the caller initializes it with a starting value, use an internal incrementing counter per pad and only update it when the report has changed
- Allow pass-through for real XInput-compatible pads
  - Currently only DS3s in SXS mode get detected and reported as connected
  - Listen to native XI APIs and pass-through any other XI pad if there is free slots

mirh, 2022-04-17:
Stupid question: if you wanted to use these extensions with the other controllers that used to benefit from SCP (Scarlet.Crush mentioned the DS2 Play.com USB Adapter, and the SpeedLink Strike FX pad ps3 controller) how would you go?

nefarius, 2022-04-17:
Well, it all depends on how/where you get the data. Since it's a proxy DLL, you can in theory "convert" any input device to it as long as it gets translated correctly. Currently I just use the well know HID API wrapper library to talk to DsHidMini, grab the raw report and translate it accordingly.

dezraj, 2022-10-10:
I'm sorry to post here but looking for "dshidmini pcsx2 qt" on Google took me to this link https://github.com/ViGEm/DsHidMini/issues/227 , which then took me here.

So, I'm trying to make pressure sensitivity work on PCSX2 Qt (latest dev build as of now) but I'm likely not being able to. I tested it with the game The Bouncer which has a pressure sensitivity test in the controller options menu, but the pressure slider just register one state (therefore, it's just digital).

I have already enabled SXS in the DSHIDMini software and placed the XInput DLL in the emulator's root folder. In PCSX2 Global Settings, I have "Enable SDL Input Source" ticked. Then I go to Controller Port 1 - DualShock 2 and select Automatic Mapping and then "SDL-0 Nefarius Software Solutions e.U. DS3 Compatible HID Device" and all buttons are mapped automatically without issues, but unfortunately the pressure sensitivity does not seem to work. I have also tried to Enable XInput Input Source but then the controller won't show in the Automatic Mapping option.

I would be very glad if you could help me on that. I have had difficulty in finding a good guide for making this work properly. Thank you so much in advance for your time.

dezraj, 2022-10-10:
> I'm sorry to post here but looking for "dshidmini pcsx2 qt" on Google took me to this link #227 , which then took me here.
>
> So, I'm trying to make pressure sensitivity work on PCSX2 Qt (latest dev [...]

Ok, I figure out what was wrong. When I set SXS mode in the DSHIDMini software I was just closing the window and then this wouldn't be really changing the option. We have to either click on "Apply changes" or unplug and plug the controller again in order that the new mode take effect. Then I could finally see the controller as an XInput source in the emulator when enabling "XInput source" in the controllers configuration.

Perhaps the tool-tip for the modes could be more clear on applying the changes in order that they take immediate effect.

Anyway, I'm sorry for the inconvenience. But I'm glad I could finally make it work. Thank you again.

PR #361 (2024-05-16): Massive SCP DLL rewrite
Resolves #213

## Changes

- Device state is now only modified once on library load, afterwards event based instead of being tied to any XI proxy calls.
- Basically rewrote everything.
- Proxying "real" XInput devices is now supported, so the DLL doesn't need to be swapped/deleted if the user simply wants to use Xbox devices with the affected game or emulator.
- Used C++ (sparingly) for better abstraction and maintenance.

## Open TODOs

- [ ] OTEL doesn't build yet
- [x] Massive amount of testing
- [x] Address some inconsistencies
  - Currently only the `XInputGetExten [... cut]
closed 2024-05-16 by nefarius