nefarius/DsHidMini #214: SCP Xinput Bridge DLL causing huge performance drop after a couple of minutes

andutrache, 2022-04-20:
- [x] I have searched open and closed issues for duplicates
- [x] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
Using the DLL to get pressure sensitivity in PCSX2 and after a couple of minutes the performance drops harshly, ultimately leading to the whole emulator becoming "Not responding".

### Steps to reproduce
- Place the DLL in PCSX2 root folder
- Enable Xinput in PCSX2 Pad settings
- Play any game

**Actual result:** Performance will start to drop
**Expected result:** Performance should stay consistent

### Screenshots

[image]

### Machine info

**CPU:** AMD Ryzen 7 2700
**CPU-Architecture:** x64
**RAM:** 16 GB
**GPU:** Nvidia GTX 1060
**Windows version:** Windows 10 Enterprise 21H1 64-bit
**Software/driver version(s):** PCSX2 v1.7.2641 (x64) and DsHidMini 2.2.282.0

### Any other helpful information

When the performance degrades the power usage of the process in the screenshot goes to Very High

###Testing / stuff i tried:
- Switching DsHidMini from SXS to Xinput and removing the DLL from PCSX2 does not cause any problems and works as expected (minus the pressure sensitivity of course) so it doesn't look to be an emulator bug
- When the slowdown occurs, disabling Xinput in the emulator and switching to Dinput makes the problem disappear (but so does the pressure sensitivity)
- Pressure sensitivity tests have been done with a padtest.elf running in the emulator
- The tests above have been done in coop with a user from the PCSX2 Discord who got the exact same result
- Ran the Legacinator program and it found no problems

###Other information:
- I imagine the performance drop will come slower or faster depending on the game being played and the power of the machine running the emulator (i used Rayman 2 Revolution in my tests)
- Have also posted the bug on the Nefarious Software Solutions discord server
- If more info or more testing is needed just ask and I will do my best to respond and help

nefarius, 2022-06-14:
Yep, that release had many lingering bugs, here's an intermediate build until the next release is ready:
- [x64](https://buildbot.vigem.org/builds/DsHidMini/master/3.0.365.0/bin/x64/XInput1_3.dll)
- [x86](https://buildbot.vigem.org/builds/DsHidMini/master/3.0.365.0/bin/x86/XInput1_3.dll)

andutrache, 2022-06-18:
Tested the intermediate build above and still get the issue, the only thing that has changed is that it takes more time for it to affect performance.

nefarius, 2022-06-18:
Darn, thanks for the feedback though!

nefarius, 2022-06-18:
Welp, once you merge the fixes it works 🤣 here you go:

- [x64](https://buildbot.vigem.org/builds/DsHidMini/master/3.0.369.0/bin/x64/XInput1_3.dll)
- [x86](https://buildbot.vigem.org/builds/DsHidMini/master/3.0.369.0/bin/x86/XInput1_3.dll)

andutrache, 2022-06-18:
Indeed it does :)

nefarius, 2022-06-28:
See #213 for further progress.

closed 2022-06-28 by nefarius