nefarius/DsHidMini #225: SCP Xinput Bridge DLL consuming a lot of CPU while playing games on Pcsx2 emulator

mcjhonns, 2022-06-17:
### Bug description
While using SCP Xinput bridge DLL on the Pcsx2 Emulator(WxWidgets or Qt version) the "Windows Driver Foundation" process starts cosuming so much CPU so the emulation performance starts to drop, it doesn't really matter if my DS3 is connected or not, this happens by just having the Moded Xinput dll on the emulator folder and enabling Xinput support. This issue keeps persisten even after the latest update of the dll posted on the previous bug report regarding this issue.
### Steps to reproduce
- Place the DLL in PCSX2 root folder
- Enable Xinput in PCSX2 Pad settings
- Play any game

**Actual result:** The performance starts to drop at the very moment of enabling Xinput
**Expected result:** The driver process consumption should be normal

### Screenshots

[image]

### Machine info

**CPU:** Intel Core i5 6600
**GPU:** Nvidia GTX 970
**CPU-Architecture:** x64
**Windows version:** Windows 10 20H2
**Software/driver version(s):** PCSX2 v1.7.2953 (x64) and DsHidMini 2.2.282.0

### Any other helpful information
I'll like to add that I tried every mode on dshidmini to be sure if this precisely happens with the SXS mode, but as I said before it happens even with the DS3 controller disconnected, I think am not missing any detail for now but I will be very pleased to give any detail that you guys ask me.

andutrache, 2022-06-18:
fixed and new build was posted here : https://github.com/ViGEm/DsHidMini/issues/214

nefarius, 2022-06-28:
See #213 for further progress.