nefarius/DsHidMini #227: A few issues when combining DualShock 3, DsHidMini and PCSX2 (the new Qt builds).

jonasos3, 2022-06-21:
DualShock 3 controller connected with USB, through DsHidMini (with the new fixed .dll). Seen as XInput source in PCSX2 Qt. Automatic binding binds everything correctly EXCEPT the two analog sticks; up/down is reversed.
It is mapped correctly when the controller is seen as a DS3 by Qt, but that makes the pressure-sensitive buttons behave as digital, defeating the purpose.

-

When trying to set the vibration motors and error shows up "No devices with vibration motors were detected.". No vibration in-game. Vibration is not working when seen as a DS3 either., although the two vibration motors are now seen. They can be mapped, but don't work.

-

To clarify; For pressure sensitive buttons to work, PCSX2 must see the input as XInput. DS3 input behaves as digital. When seen as XInput, the axis up/down are reversed (easily fixed by mapping them myself of course). 
The second issue with vibration; XInput; the motors don't show up in the mapping at all. In DS3 input they can be mapped, but don't work in PCSX2. 
Everything works fine in RPCS3. 
This applies when the DualShock 3 in DsHidMini is set to any of the modes; SDF, SXS, DInput generic etc.

Windows 10 x64.

nefarius, 2022-06-21:
> When trying to set the vibration motors and error shows up "No devices with vibration motors were detected.". No vibration in-game. Vibration is not working when seen as a DS3 either., although the two vibration motors are now seen. They can be mapped, but don't work.

Yeah I have a hunch on what's wrong there, will check.

nefarius, 2022-06-21:
[Here you go](https://buildbot.vigem.org/builds/DsHidMini/master/3.0.371.0/bin/x64/XInput1_3.dll).

jonasos3, 2022-06-21:
Fantastic, thanks!

nefarius, 2022-06-28:
See #213 for further progress.

closed 2022-06-28 by nefarius