nefarius/DsHidMini #228: PLAYSTATION(R)3Conteroller - DS3 - STATUS_DEVICE_POWER_FAILURE - code 10 - C000009E

camnop, 2022-06-23:
Sony Dualshock 3 - PLAYSTATION(R)3Conteroller shows STATUS_DEVICE_POWER_FAILURE - code 10 - C000009E

I tried resetting the PS3 controller and I have already rebooted the computer, but it still remains the code 10.

Driver: dshidmini v2.2.282.0
Microsoft Windows 11 - 10.0.22000 (X64)
CPU: 11th Gen Intel(R) Core(TM) i7-11800H

[image]
[image]

```
DS3 Compatible HID Device
USB\VID_054C&PID_0268\5&2A83CFE&0&4
USB\VID_054C&PID_0268&REV_0100
01806400
DN_HAS_PROBLEM
DN_DISABLEABLE
DN_REMOVABLE
DN_NT_ENUMERATOR
DN_NT_DRIVER
C000009E
```

**Device USB\VID_054C&PID_0268\5&2a83cfe&0&4 configured.**

```
Driver name: oem253.inf
Class Guid : {2b959c9d-a4ed-4464-809d-0f1118cf055d}
Driver date: 11/08/2021
Driver version: 2.2.282.0
Driver provider: Nefarius Software Solutions e.U.
Driver section: dshidmini.USB.NT
Driver classification: 0xFF0001
ID: USB\VID_054C&PID_0268
Drivers: input.inf:USB\Class_03:00FF3202
Father device: USB\ROOT_HUB30\4&1fa01397&0&0
```

**Device USB\VID_054C&PID_0268\5&2a83cfe&0&4 had a problem to start.**

```
Driver name: oem253.inf
Class Guid : {2b959c9d-a4ed-4464-809d-0f1118cf055d}
Service: mshidumdf
Filters: WUDFRd nssmkig
Problem: 0xA
Problem Status: 0xC000009E
```

inaqi69, 2022-07-24:
I am facing nearly same issue like this i have posted at forums below:
https://forums.vigem.org/topic/651/ps3-controller-blinking-all-four-lights-fast-when-pressing-ps-button-and-then-turns-off-also-no-lights-when-connecting-through-usb-device-manager-shows-ds3-compatible-hid-device-yellow-triangle

nefarius, 2022-08-01:
We most probably need a trace, will sketch together a small guide on how to do so.

nefarius, 2022-08-22:
Related to #244