nefarius/DsHidMini #232: Dualshock 3 disconnected but not really

arcanechocolate, 2022-08-01:
Let's begin with a checklist: Replace the empty checkboxes [ ] below with checked ones [x] accordingly. -->

- [x] I have searched open and closed issues for duplicates
- [ ] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
This is ridiculously bizarre.
Basically, whenever I'm playing MGS3 HD on RPCS3 and get hit by The Fear boss, the Dualshock 3 on windows/rpcs3/etc loses all of its functionality apart from vibration. It's still recognized by windows and software, but the buttons/analogs stop working altogether. Nevertheless, you can still send vibration signals to it through applications. I can't even fathom the reasoning for this and how specific it seems to be.

Obviously, the hit from that boss triggers RPCS3 to send a signal to the controller, probably a vibration signal, and for some reason that signal is breaking every other functionality from the controller apart from vibration. This is my assumption. I am playing wired, might I add.

### Steps to reproduce
Play MGS3 HD, get hit by a dart from The Fear while in first-person.

**Actual result:** Controller loses all functionality apart from vibration.
**Expected result:** Controller shouldn't lose any functionality.

### Screenshots

### Machine info
**CPU-Architecture:** i5-10600K
**Windows version:** Windows 11 22000
**Software/driver version(s):** 
DSHIDMINI 2.2.282.0
RPCS3 0.0.23-13988-052725fd Alpha | master

### Any other helpful information

nefarius, 2022-08-01:
Wired or wireless?

[Try playing around with these settings](https://vigem.org/projects/DsHidMini/Output-Rate-Control-Explained/).

Cheers

nefarius, 2022-08-01:
> I mentioned wired in the description.

Indeed, my bad.

> I've also tried different values in the app. Enabled/disabled output report rate control, duplication, and different values for the rate control (from 50 to 300ms). Applied and restarted app after each change.

Hm, interesting case. Let me think about it.