nefarius/DsHidMini #234: - Right Rumble Motor only turns ON or OFF / Controller indicator LED is randomly selected -

Schwarzdrossel77, 2022-08-03:
Let's begin with a checklist: Replace the empty checkboxes [ ] below with checked ones [x] accordingly. -->

- [X] I have searched open and closed issues for duplicates
- [X] I am submitting a bug report for existing functionality that does not work as intended
- [X] I have read https://vigem.org/Community-Support/
- [X] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
1. 
Hello! First of all a big thank you for this wonderful Software! It works really well and i didn't encounter any big issues so far.
But there are two things that don't seem to work quiet right.
One of them would be that the indicator LED on my Official Dualshock 3 Controller gets randomly selected, when i restart my PC.
What i mean by that is that everytime i restart my PC and reconnect the Controller via Bluetooth or Cable, one random indicator LED from 1-4 lights up, even though there where never any other Controllers connected. Shouldn't it always be numba one?

2.
The other thing which is a bit annoying is that the right Rumble Motor seems to be an ON/OFF Switch. I noticed it first when i tried to play Dirt Rally 2. To confirm the issue i downloaded this simple Program called ControllerVibrationTester_v1.3 (In this YouTube Video is a Download link: https://www.youtube.com/watch?v=TyGZQv3ybTI). With this Tool i tested the Vibration Motors on my Official Dualshock 3 Controller. I can control the speed of the left Rumble Motor perfectly fine. The right one on the other hand only turns ON or OFF. Is it supposed to be that way?

### Steps to reproduce

1.
- Turn on PC
- Connect Official Dualshock 3 Controller via Bluetooth/Cable

**Actual result:** Random indicator LED lights up.
**Expected result:** Indicator LED 1 lights up (because there is only one Controller connected).

2.
- Turn on PC
- Download and install ControllerVibrationTester_v1.3 from YouTube Video (https://www.youtube.com/watch?v=TyGZQv3ybTI)
- Connect Official Dualshock 3 Controller via Bluetooth/Cable
- Test the Rumble Motors of the Controller with the Tool

**Actual result:** If the slider percentage for the right Rumble Motor is above 1%, it starts spinning at full speed.
**Expected result:** You can precisely control the speed of the right Rumble Motor with the Slider.

### Machine info
**CPU-Architecture:** x64
**Windows version:** Windows 10 21H2
**Software/driver version(s):** DsHidMini Driver v2.2.282 / BthPS3 Bluetooth Drivers v2.1.153

### Any other helpful information
I have multiple Devices connected to my Bluetooth Dongle. (Asus BT-400).

With Kind Regards,
Schwarzdrossel

Kanuan, 2022-08-03:
> the indicator LED on my Official Dualshock 3 Controller gets randomly selected

It is explained on [DsHidMini README's feature section](https://github.com/ViGEm/DsHidMini#features) that the LED represents the controller current battery level. DsHidMini can't currently show which "slot" the controller is into because either it does not make sense depending on the current HID Device Mode or some other limitation. Even ScpToolKit's "LEDs show current slot" was not actually reliable.

-----------

> the right Rumble Motor seems to be an ON/OFF Switch.

The DS3 controller's right/light/small rumble motor can't have its strength adjusted, it's only ON or OFF. It's not an issue per-se, just a hardware limitation.

A workaround for this limitation has already been developed and implemented on DsHidMini's source code. What the workaround does is explained on Issue #51 , but the TL;DR is there will be an option in the future to use the Left motor for both the heavy and light rumble commands. It's not perfect but I've run some tests with the workaround and it's miles better than just having the annoying small rumble toggle, specially in racing games. Users will lose the left/right difference but the overall rumble experience, even for minor effects like the character steps, will be maintained when possible.

Though the workaround already is implemented in the code, the next version of DsHidMini will require a full replacement of the current configuration interface and so there is no ETA for the next version. It will be available one day in the future.

nefarius, 2022-08-03:
What @Kanuan said 😉 In summary:

> Shouldn't it always be numba one?

README 😛 

[image]

> The right one on the other hand only turns ON or OFF. Is it supposed to be that way?

Yes, the SIXAXIS/DS3 **hardware** has no adjustable strength. v3 will offer more customization and workarounds as @Kanuan described.

closed 2022-08-03 by nefarius