nefarius/DsHidMini #252: DSU Client support

hutiwephy, 2022-09-16:
Can we get support for DSU Client it would help a lot with Gyro input on Cemu and Ryujinx.

I currently have to use the DsHidMini to feed a fake DS4 to DS4windows so I can use it on Cemu.

Having a DSU server on DsHidMini would help a lot!

hutiwephy, 2022-09-16:
PS: when using the PS3 Controller over Bluetooth as a DS4windows device it is not picked up by DS4windows, but when connect by USB it is recognized

the lights when connected via USB change, but not when connected via Bluetooth

Kanuan, 2022-09-16:
Dshidmini doesn't support gyro/motion in a way that can be used with modern emulators.

The main developer has no interest nor time currently to try translating DS3 gyro into the currently used cemuhook format (not sure if this is the current name).

The only emulator that can read the DS3 motion sensors is RPCS3 and basically because it is doing all the work itself since it's a PS3 emulator to begin with.

Unless you have another use case for a DSU client that doesn't involve gyro there is no benefit from what I know that could be gained by developing one

hutiwephy, 2022-09-16:
@Kanuan in that case can we focus on the Bluetooth issue or should I drop it or move it somewhere else?

Kanuan, 2022-09-16:
On the vigem.org website check the community section on how to properly seek support. You can create a DISCUSSION thread in the discussion tab too, but via discord it would be easier.

Won't be able to help anymore for today though

nefarius, 2022-09-16:
As @Kanuan correctly summarised, this most probably requires logic that properly transforms the DS3 gyro data, which we do not have. Contributions welcome. Also see #217

nefarius, 2022-09-17:
> @Kanuan in that case can we focus on the Bluetooth issue or should I drop it or move it somewhere else?

Discord is preferred, it usually helps to have instant chat capabilities and more people available that can help in different time zones.

retrofan2019, 2023-02-17:
> 

so the motion controls work on cemu and ryujinx if the controller is set to a fake ds4 on ds4windows ... if that is the case then i'll get rid of ScpToolkit .. I wanted to update to DsHidMini for a long time but the motion controls thing was the only reason i didn't go for it .

nefarius, 2023-02-17:
> > 
> 
> so the motion controls work on cemu and ryujinx if the controller is set to a fake ds4 on ds4windows ... if that is the case then i'll get rid of ScpToolkit .. I wanted to update to DsHidMini for a long time but the motion controls thing was the only reason i didn't go for it .

Nothing new came up since the last post, see #217

retrofan2019, 2023-02-17:
> Nothing new came up since the last post, see #217

yep i saw that and understand that DsHidMini doesn't support motion control yet ( hopefully it will be possible in the future) ... it's just the user who started the thread said he can't directly use the motion controls but it's possible through ds4windows ... if that is the case i will update right now to DsHidMini .. i really want to get rid of scp toolkit and all the headache that comes with it ... and i have been following the amazing work done here and wanted to update for a really long time .

nefarius, 2023-02-17:
> yep i saw that and understand that DsHidMini doesn't support motion control yet ( hopefully it will be possible in the future) ... it's just the user who started the thread said he can't directly [...]

Ah now I understand what you meant, thanks for clarifying 👍

sunnyqeen, 2023-03-04:
DsHidMini do have motion data in SXS mode, via FeatureReport which used by RPCS3
I have added new DS3 native support (SXS mode) to DS4Windows, which supports motion data output.
see https://github.com/Ryochan7/DS4Windows/pull/2850
and https://github.com/Ryochan7/DS4Windows/pull/2852

retrofan2019, 2023-03-05:
> DsHidMini do have motion data in SXS mode, via FeatureReport which used by RPCS3 I have added new DS3 native support (SXS mode) to DS4Windows, which supports motion data output. see [Ryochan7/DS4Windows#2850](https://github.com/Ryochan7/DS4Windows/pull/2850) and [Ryochan7/DS4Windows#2852](https://github.com/Ryochan7/DS4Windows/pull/2852)

great work! .. looking forward to test it once it gets released ... finally i'll be able to move from scp toolkit to dshidmini and ds4windows .

nefarius, 2023-03-05:
Very nice, that should be good material to someday come up with a more direct way to integrate it into DSHM core 👍

retrofan2019, 2023-03-05:
> DsHidMini do have motion data in SXS mode, via FeatureReport which used by RPCS3 I have added new DS3 native support (SXS mode) to DS4Windows, which supports motion data output. see [...]

I just noticed that not only the motion data is added , there are a few Bluetooth fixes ... is the ps3 controller now able to connect through Bluetooth in ds3windows ?

nefarius, 2023-03-05:
> > DsHidMini do have motion data in SXS mode, via FeatureReport which used by RPCS3 I have added new DS3 native support (SXS mode) to DS4Windows, which supports motion data output. see [Ryochan7/DS4Windows#2850](https://github.com/Ryochan7/DS4Windows/pull/2850) and [Ryochan7/DS4Windows#2852](https://github.com/Ryochan7/DS4Windows/pull/2852)
> 
> I just noticed that not only the motion data is added , there are a few Bluetooth fixes ... is the ps3 controller now able to connect through Bluetooth in ds4windows ?

This has been possible since forever when BthPS3 + DsHidMini support was first added like 3 years ago or so?

retrofan2019, 2023-03-05:
> 

oh yes i know ... it just doesn't work in ps4windows mode ... only usb.

nefarius, 2023-03-05:
> > 
> 
> oh yes i know ... it just doesn't work in ps4windows mode ... only usb.

... yes it does? Why wouldn't it.

retrofan2019, 2023-03-05:
> 
it is one of the things that doesn't work well in ps4windows mode ..check under what does not work  in ps4windows mode
https://vigem.org/projects/DsHidMini/DS4-Mode-User-Guide/

and I actually can confirm that it doesn't show up as Bluetooth and some options won't work on ps4windows side,  because after the news that ps4windows will have native motion support ,I just went ahead and removed  scp toolkit and installed BthPS3 + DsHidMini  .. and they work perfectly fine ,really easy to set up and way better than scp toolkit and now my Bluetooth device is free lol ..but yea in ps4windows mode it only appears as USB ..it's a really minor issue since it works perfectly fine on DsHidMinis side .

nefarius, 2023-03-05:
Ah now I get what you mean, yeah there isn't really a way around that limitation.

retrofan2019, 2023-03-08:
@sunnyqeen 

the new release is out with the SXS mode support ,great work!... if i set the controller to sxs mode in DsHidMini , now it appears in ds4windows ... but only in usb , it just didn't get detected in Bluetooth mode ..

and after enabling the udp server , i actually got motion control in cemu ... but sadly it's not functioning right ... only up and down movement r detected and it's slow ... i even used the padtest (by rajkosto the creator of cemuhook ) and yea i can see there is a problem as the movements r correctly detected but it's really slow ... i move to  a certain position and it is correctly detected but takes a few seconds to get there .

sunnyqeen, 2023-03-08:
Bluetooth is controlled by DsHidMin, not the application.
 
PS3 controller has limitation for the gyro, only single yaw. (left/right rotation)
I have 2 of them, only the old module has good response for yaw (shift 500 unit), the other A1 module is very bad, can only shift 1 unit (better than nothing).

The accel values are also not so accuracy, but work better than gyro.
 
So to make it movements fast, you need to scale them.

retrofan2019, 2023-03-08:
> Bluetooth is controlled by DsHidMin, not the application.
>
> PS3 controller has limitation for the gyro, only single yaw. (left/right rotation) I have 2 of them, only the old module has good response [...]

I see .. well, the bluetooth issue is not a big deal right now ...getting decent gyro controls is the main issue ..

and yep i know that the ps3 gyro is not good to begin with and limited .. it was never perfect on cemu ..but it did work and all the controllers i had worked decently using cemuhook and the patched dlls for scptoolkit .. but here it's really slow a..i don't know if u tested it with padtest and cemu ... but it's almost not moving.

and is there an option to be able to scale them and get faster movement ?

retrofan2019, 2023-03-10:
@sunnyqeen 

so i ended up digging all my ps3 controllers and fixed some normal issues like rumble not working or sticky buttons and ended up with 5 perfectly working ps3 controllers .. and i tested all of them and i think i see what u mean by old version having a good yaw response ..cuz only 1 controller actually worked really well with x rotation (i assume it's one of the original ones i got with the ps3) but it still had slow movement up and down ...

and i tested the controller directly with sxs mode in DsHidMini and then through ds4windows sxs mode on rpcs3 ..and i seem to get the same outcome ...it runs really well on rpcs3 , and they even have control led for player number or battery  ..

so i don't know if there is some extra code on rpc3 side that is still missing that could help fix the issue with unresponsive and very slow motion controls on cemu and padtest using the udp server (or basically anything outside rpcs3) .. 

but i did dig up the code used by Rajko (creator of cemuhook and padtest) on scptoolkit to make proper calibration of the ds3 gyro and released it as a patch for  v1.7.277.16103 scptoolkit ..one thing i know for sure it worked perfectly fine on my ps3 controllers.

[](https://github.com/nefarius/ScpToolkit/compare/master...rajkosto:ScpToolkit:master)

maybe that could be helpful to find out what is causing the problem ... it seems like he needed to tweak things to make it work like yaw inversion for certain models ..and swapping y and z accel  (actually i can see from the readings on rpcs3 vs ps4windows that they r swapped)

sunnyqeen, 2023-03-10:
The code I have done is following rpcs3 DS3 pad code .
I checked a bit Scptoolkit, I think they use a binary ds3cal.dll, which has no source code.
we have no pitch value (always 0), that why it's slow up and down.
For the X Y Z accel, you should trust the padtest, if you make upside down of your controller, the picture should do the same.

nefarius, 2023-03-10:
ds3cal.dll is a recompile of parts of the Sony PS3 firmware and therefore can not be open sourced nor used in DsHidMini without breaking the law, so that's neat that it exists but unfortunately completely useless for this project.

nefarius, 2023-03-10:
The ScpToolkit with the calibration stuff is a fork, not my work, I just relay info I've learned and gathered over time about it.

retrofan2019, 2023-03-10:
> 

oh ... I guess since ur name was all over the app so i didn't even check .. but  u definitely know a lot of about it and did some work on it .. so u know the parts that can't be used like the sony dll  .. but like i said i just thought the part of the code we can access might be helpful ... because Rajko definitely managed to get proper gyro for the ds3, so it can be done .

nefarius, 2023-03-10:
Oh I know it can be done, simply not by me. If you wanna see proper motion support directly within DsHidMini, contribute some C code that takes care of calibration and transformation. That's it. Until then there's nothing more for me to add to the topic.

nefarius, 2023-03-10:
The "you" is directed towards anybody reading this thread, not at you as a single person.

eVenent, 2023-03-11:
Oh wow, great job @sunnyqeen! I see that we are close to have proper support for DS3 motion controls directly in DsHidMini. I hope we will achieve that sooner or later. Your update for DS4Windows woke up DS3 users and it's soo good feeling, that after years we can have support for this great controller. It's another good news this year, after adding [support in SDL2 - not windows yet.](https://github.com/libsdl-org/SDL/issues/5148) 

As user I am waiting for good motion controls support for DS3 (gInput, DSU and DS4 gyro emulation) and proper SXS mode/official driver support for pressure-sensitive buttons in new PCSX2. If as community we will achieve it, then DS3 can be officially announced as fully revived controller. ❤️ Thank you guys, that you still care.

I don't know if it will help, but RetroFighters Defender Bluetooth Edition controller has DS3 mode and is designed to work on PS3 and is fully compatible with console (pressure-sensitive buttons, motion controls etc). Maybe newer hardware will help with developing proper DSU support, as they somehow recreated logic used in DS3. Good luck with further development. 🤞

nefarius, 2023-03-11:
IMHO so far the Retrofighters Defender offers pressure sensitivity but no motion sensors, correct me if I'm wrong.

eVenent, 2023-03-11:
You're right about basic Defender. New Bluetooth Edition implemented motion controls to be a solid alternative for DualShock 3.

nefarius, 2023-03-11:
> You're right about basic Defender. New Bluetooth Edition implemented motion controls to be a solid alternative for DualShock 3.

Great, good to know!